WatershedArts/Footfall #3: No makefile

anthgur, 2015-10-07:
I installed openFrameworks, ran `getrepos.sh` (thanks for including that!), but it appears that there isn't a makefile in the `RPi` directory. Is there a step that I missed/how do I configure and build the project?

I copied over the makefile from an example project but make failed around here:
```
Compiling RPi for Release
make[1]: Entering directory '/home/pi/openFrameworks/apps/myApps/RPi'
... Skipped for brevity...
Linking bin/RPi for linuxarmv7l
... Skipped for brevity...
collect2: ld returned 1 exit status
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:296: recipe for target 'bin/RPi' failed
make[1]: *** [bin/RPi] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/RPi'
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:126: recipe for target 'Release' failed
```

DHaylock, 2015-10-07:
Hi @anthgur, yeah there is no make in the repo. You might have to copy the src into a new project folder and alter the addon.make file to include the necessary addons.

bsamaroo, 2015-10-09:
Thanks for uploading this, very excited to get it working. I am having the same problem as @anthgur. I copied the files within the RPi directory into a new project, but am unsure what I should be adding to the addons.make file to make it work properly. Would appreciate your help on this. Thanks

DHaylock, 2015-10-09:
Hi @anthgur i've added the make files, one thing to ask is have you put the RPI folder inside your oF apps/myApps directory?

closed 2015-10-15 by DHaylock

anthgur, 2015-10-16:
Hey @DHaylock, as a follow up I got this working. I think it was a problem with my build of  openframeworks. Thanks again.

DHaylock, 2015-10-16:
No worries @anthgur