WatershedArts/Footfall #4: Error sending events to upload.php

bsamaroo, 2015-10-09:
Hi there, thanks for sharing this code. I am a beginner and having some trouble getting the events to send to the upload.php which is sitting on my PHP/SQL compatible server. When an event occurs, I get an error saying "Access denied for user '<Your-Database-N @ localhost' (using password: YES).

I edited the config.xml to point to the correct URL as you had indicated in your instructions. I am unsure what to put for "secretkey".

I am also wondering where exactly I should enter my mySQL username into the code so that the server knows where to put the information (and whether a password is required).

Any help would be much appreciated. Thanks

DHaylock, 2015-10-09:
Hi @bsamaroo, 

Have you put your database name and password in the dbconnect.php file?

Right so in the includes/secretkey.php, you have the following code.

```<?php
$secret = "Secret Key";
if (!isset($_POST['secret']) || $_POST['secret'] != $secret) {
	header('HTTP/1.0 403 Forbidden');
	exit;
}
?>
```
The secret key can be a completely randomised string, as long as you place the same secret key into the config.xml it should upload. Its an extra security method to ensure people cant post nonsense to your scripts.

bsamaroo, 2015-10-09:
Thanks @DHaylock, that is very helpful. 

I have updated the secret key in my secretkey.php and config.xml files, and added my mysql username/password to the dbconnect.php file.

For "host" in the dbconnect.php file, what should I put as the $host? Should this be the IP address of my raspberry pi?  Thanks again for your help.

oliverhumpage, 2015-10-09:
Hi,

The $host would be the host that your mysql server is running on. On a normal scale setup this would just be “localhost”.

However, I’ve just noticed that there isn’t a database schema in the repo, so there’ll be no table for your footfall events to be inserted into yet! Sorry.

I’ll take a look at that and try to get one uploaded. For now, go to your database (command line MySQL or phpMyAdmin, your choice) and use the following SQL to create the table “data” in which your footfall data will be stored:

CREATE TABLE IF NOT EXISTS `data` (
  `timestamp` timestamp NOT NULL DEFAULT CURRENT_TIMESTAMP,
  `locationID` tinyint(4) NOT NULL,
  `event` tinyint(4) NOT NULL,
  KEY `timestamp` (`timestamp`)
) ENGINE=InnoDB DEFAULT CHARSET=utf8;

If you need any help setting up your database and/or its tables/logins, Google or stackoverflow will probably be more helpful than any quick email I can dash off :)

Cheers,

Oliver (colleague of David).

bsamaroo, 2015-10-09:
@oliverhumpage 

Thanks very much for your response. I will try out your advice above and will look forward to an update of the repo if you get around to it! Thanks for all of your work on this.

bsamaroo, 2015-10-10:
@oliverhumpage would be very grateful if you could help out with one more question. I have inputted the $dbname, $passkey and $host into dbconnect.php but I am getting an error on the raspberrypi when the camera detects a blob and the raspberrypi  tries to send the information to the database. The error says:

SQLSTATE[28000] [1045] Access denied for user 'a1234567_test'@'10.1.1.24' (using password: YES)

Any ideas on how to fix this? My database is called a1234567_test and has a user named a1234567_user, but the dbconnect.php file does not ask for a username or password of the database user. 

Thanks again!

bsamaroo, 2015-10-10:
Hi again @DHaylock and @oliverhumpage. Thanks for all of your help so far. 

My raspberry pi is now successfully sending data to the server (phpMyAdmin indicates that records are being added to the table. 

However, when I open events.php the figures are still sitting at 0. I updated main-events.js to point to the location of my upload.php file. I have downloaded chart.js and unzipped the files into the same directory the footfall files are located on my server. Any thoughts on this?

Thanks again for all of the assistance.

bsamaroo, 2015-10-10:
I just noticed that all of the data in phpMyAdmin is showing up with locationID=0 and event=0, so I'm guessing that is the issue. My XML file has locationID set to 1 and I haven't changed anything related to the event value.

closed 2015-10-11 by bsamaroo

rrrikin, 2016-01-27:
Hey , 
I am having similar issue that you described on last post. That all I see is 0s in time stamp, event and LocationID on mySQL db table. Any thoughts what i am missing? I am a beginner and building this up for a non-profit org.
Thank you

DHaylock, 2016-01-28:
Hi @rrrikin,

No worries, Have just pushed some changes that should resolve the issue. 

What it is, is that when you use prepared statements you don't need to use my_real_escape_string as PDO will escape the variables.

Best