WatershedArts/Footfall #5: View Footfall window through VNC

elisiariocouto, 2015-10-15:
I've successfully compiled and ran the project on my RPi 2 but I'm trying to get a way to view the app window through VNC. Currently, my RPi is not connected to any video device. I've tried tightvnc with no success and I'm now using x11vnc which has the ability to show the same display as the HDMI port (again, which is not connected to anything). I have a working VNC connection on display :0 but I've tried to run the app with DISPLAY=:0, DISPLAY=:1, DISPLAY=:0.0 but the window never shows up. Do you have any idea why?
Thanks in advance

oliverhumpage, 2015-10-15:
There’s no app window on the pi. The command-line daemon on the pi just sends footfall events to a remote web server (which you also have to set up), and that saves them to a database. You then connect to that web server to view the charts.

In theory you could run that web server on the pi itself, but either way you view data through a browser, not VNC.

closed 2015-10-15 by oliverhumpage

elisiariocouto, 2015-10-15:
I'm not talking about the data, I'm talking about the window that shows what the camera is "seeing"... I found the problem. This app creates an OpenGL ES window, which most VNC servers can not display because they aren't X windows. I'll try to use VirtualGL to forward it through VNC.

fparisk, 2016-03-04:
How did you guys are remotely executing the app? 
I have try everything and I am not able to run it... any suggestions?

DHaylock, 2016-03-04:
Hi @fparisk 

We set the app running using [crontab](https://help.ubuntu.com/community/CronHowto) 
`crontab -e`

So at what ever time you want to start the app you do the following.

```
00  07  *   *   *   /home/pi/your-app-folder/bin/Footfall
00  00  *   *   *   /sBin/reboot
```

Alternatively, you could run it when the pi boots by adding the following in to the /etc/rc.local
`/home/pi/your-app-folder/bin/Footfall`

or 

If you want to manually trigger
`sudo DISPLAY=:0  /home/pi/your-app-folder/bin/Footfall`

fparisk, 2016-03-08:
Hi @DHaylock,

I was able to execute it through ssh session by using sudo DISPLAY=:0 /home/pi/your-app-folder/bin/Footfall. Later I will try crontab.

Thanks for your help

mgutierrez80, 2016-06-02:
@elisiariocouto Hi, were you able to view something using your method?

elisiariocouto, 2016-06-03:
@mgutierrez80 I was able to forward the session using this VNC server: https://github.com/hanzelpeter/dispmanx_vnc

mgutierrez80, 2016-06-03:
it works! thanks @elisiariocouto