WatershedArts/Footfall #6: Adjusting for height of camera

bsamaroo, 2015-10-23:
Hi, 

I have mounted my raspberrypi and the software is running great. The only issue is that instead of logging one person walking in and out of the store, it often logs 2-3. I assume that my ceiling is lower than yours was. 

I was hoping you could provide some guidance on how to adjust the config.xml file to compensate for the lower ceiling height.

Thanks again.

oliverhumpage, 2015-10-23:
In general the *BLOBWIDTH settings are the ones to edit. Because people walking shoulder-to-shoulder tend to appear as a single blog, we set the app to detect abnormally wide blobs and count them as 2 people (midblobwidth) or 3 people (maxblobwidth).

However, do also be aware that lighting plays a significant part in this: sometimes shadows can count as a separate person. The only real way around that is to use a depth sensor like a kinect, rather than a plain 2D camera. Or some much fancier algorithms. Either way, we had to mess around with careful placement for a while before finding one that was decently accurate.

Oliver.

bsamaroo, 2015-10-25:
@oliverhumpage thanks for this response. 

Hoping you could provide a little information on the variables "history", "mogthreshold", "maxdistance", "persistance", "minarea", "maxarea", "contourthreshold" and "threshold". Some of these are mentioned in your blog post but I would really appreciate if you could expand a bit on these.

Thanks again.

DHaylock, 2015-10-25:
Hi @bsamaroo, So:

"history" : is the length of frames the MOG2 holds to adapt to the background. this is filled continuously!
"mogthreshold" : is the threshold to describe the background model
"maxdistance" : is the furthest distance a blob or person can travel in a set number of frames
"persistance" : is how long you want to track the blob before it starts being deleted
"minarea" : is the smallest a blob can be, this is the radius of the blob. Anything below this will not be recognised.
"maxarea": again is the largest a blob can be.
"contourthreshold" : is the threshold of the blob detection, so if the blob is white a value of around 200 is ok .
"threshold" : is the generic threshold of the image

closed 2016-01-28 by DHaylock