WatershedArts/Footfall #7: compile crashing on ofxCvPiCam

d0ubleI, 2015-12-22:
Hello,
I've run into a strange issue while compiling. I have completely rebuilt my pi just to run this app and reinstalled OF. I was able to compile another project so I'm pretty sure OF is installed properly. Just running into the issue below during the build: 

./../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:3   60: recipe for target '../../../addons/obj/linuxarmv6l/Release/ofxCvPiCam/libs/m   mal/src/interface/mmal/core/mmal_port_clock.o' failed
make[1]: *** [../../../addons/obj/linuxarmv6l/Release/ofxCvPiCam/libs/mmal/src/i   nterface/mmal/core/mmal_port_clock.o] Error 1
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:3   60: recipe for target '../../../addons/obj/linuxarmv6l/Release/ofxCvPiCam/libs/m   mal/src/interface/mmal/util/mmal_graph.o' failed
make[1]: *** [../../../addons/obj/linuxarmv6l/Release/ofxCvPiCam/libs/mmal/src/i   nterface/mmal/util/mmal_graph.o] Error 1
In file included from ../../../addons/ofxCvPiCam/libs/mmal/src/interface/mmal/co   re/mmal_port.c:31:0:
/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src/interface/mmal/core/mmal   _port_private.h: At top level:
/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src/interface/mmal/core/mmal   _port_private.h:93:67: error: unknown type name ‘MMAL_CLOCK_PAYLOAD_T’
 typedef void (*MMAL_PORT_CLOCK_EVENT_CB)(MMAL_PORT_T *port, const MMAL_CLOCK_PA   YLOAD_T *event);
                                                                   ^
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:3   60: recipe for target '../../../addons/obj/linuxarmv6l/Release/ofxCvPiCam/libs/m   mal/src/interface/mmal/core/mmal_port.o' failed
make[1]: *** [../../../addons/obj/linuxarmv6l/Release/ofxCvPiCam/libs/mmal/src/i   nterface/mmal/core/mmal_port.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/sensor'
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:1   26: recipe for target 'Release' failed
make: *** [Release] Error 2

If you have any ideas of what I can try to fix this i would be very great-full. Let me know if you need any more information about the rest of the output or my device. its a ARM6 pi as you can probably tell

stewiem2000, 2015-12-22:
Hello,
Googling brought me to this page: https://github.com/orgicus/ofxCvPiCam/issues/3
Perhaps you're experiencing the same library conflict (ofxCvPiCam vs Jessie's)?
Stewart.

d0ubleI, 2015-12-23:
I tried that right before your post. Defiantly got it further through compilation. However, now im getting a new error:

fApp.cpp:(.text+0xec4): undefined reference to `ofxXmlSettings::getAttribute(std::string const&, std::string const&, int, int)'
ofApp.cpp:(.text+0xf5c): undefined reference to `ofxXmlSettings::getAttribute(std::string const&, std::string const&, int, int)'
ofApp.cpp:(.text+0xf9c): undefined reference to `ofxXmlSettings::popTag()'
ofApp.cpp:(.text+0xfa4): undefined reference to `ofxXmlSettings::popTag()'
obj/linuxarmv6l/Release/src/ofApp.o: In function `ofApp::~ofApp()':
ofApp.cpp:(.text._ZN5ofAppD2Ev[_ZN5ofAppD5Ev]+0x20c): undefined reference to `ofxXmlSettings::~ofxXmlSettings()'
collect2: error: ld returned 1 exit status

I tried googleing this one but did not get any valuable info... going to keep working on it but if anyone has any ideas it would be fantastic. 

Has anyone complied this on a ARM6 pi? Just making sure that has nothing to do with it..

d0ubleI, 2016-01-04:
I actually bought a brand new Pi to test this out. Still getting the same general error when compiling:

In file included from ../../../addons/ofxCvPiCam/libs/mmal/src/interface/mmal/util/mmal_graph.c:32:0:
/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src/interface/mmal/core/mmal_port_private.h: At top level:
/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src/interface/mmal/core/mmal_port_private.h:93:67: error: unknown type name ‘MMAL_CLOCK_PAYLOAD_T’
 typedef void (*MMAL_PORT_CLOCK_EVENT_CB)(MMAL_PORT_T *port, const MMAL_CLOCK_PAYLOAD_T *event);
                                                                   ^
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:360: recipe for target '../../../addons/obj/linuxarmv7l/Release/ofxCvPiCam/libs/mmal/src/interface/mmal/util/mmal_graph.o' failed
make[1]: *** [../../../addons/obj/linuxarmv7l/Release/ofxCvPiCam/libs/mmal/src/interface/mmal/util/mmal_graph.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/sense'
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:126: recipe for target 'Release' failed
make: *** [Release] Error 2

Any one have any ideas on this? Or would anyone be willing to share a working image of this that i can take a look at? Thanks so much!

DHaylock, 2016-01-05:
Hi @d0ubleI, which RPi are you compiling on? 
I'm seeing linuxarm7l, it was originally built for arm6l.

Going to see if i can find out why its not compiling

David

d0ubleI, 2016-01-05:
Hi @DHaylock DHaylock,
Attempting on 71 now but previously tried on an earlier pi. If you could figure out why its not compiling for me that would be great. I'm also going to engage some other developers on this. We are setting up a tracking device as a test and this open CV enabled viewer would be so useful. 

If anyone has a image of a pi running a dev version of this and could upload it that would be super helpful too, I totally understand if that is not possible though.
-ian

hexagon-robotics, 2016-01-19:
Hello,

Anything new about compiling on Raspberry Pi 2 with 4.1.15-v7+ # 830 ?

I have installed OF from master branch. And get the following errors

Best regards,
--javier

```pi@raspberrypi:~/openFrameworks/apps/devApps/RPi $ sudo make
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libpulse-simple alsa gtk+-3.0 libmpg123 
Compiling OF library for Release
make[1]: Entering directory '/home/pi/openFrameworks/libs/openFrameworksCompiled/project'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libpulse-simple alsa gtk+-3.0 libmpg123 
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libpulse-simple alsa gtk+-3.0 libmpg123 
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libpulse-simple alsa gtk+-3.0 libmpg123 
Done!
make[1]: Leaving directory '/home/pi/openFrameworks/libs/openFrameworksCompiled/project'

Compiling RPi for Release
make[1]: Entering directory '/home/pi/openFrameworks/apps/devApps/RPi'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libpulse-simple alsa gtk+-3.0 libmpg123 
g++ -c -O3 -DNDEBUG -Wall -std=c++14 -DGCC_HAS_REGEX -march=armv6 -mfpu=vfp -mfloat-abi=hard -fPIC -ftree-vectorize -Wno-psabi -pipe -DOF_USING_GTK -DOF_USING_GTK -DOF_USING_MPG123 -DTARGET_RASPBERRY_PI -DSTANDALONE -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -D_FORTIFY_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM  -I/opt/vc/include -I/opt/vc/include/IL -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -D_REENTRANT -pthread -I/usr/local/include/opencv -I/usr/local/include -I/usr/include/gstreamer-1.0 -I/usr/include/AL -I/usr/include/alsa -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/arm-linux-gnueabihf/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng12 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng12 -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -I/home/pi/openFrameworks/libs/fmodex/include -I/home/pi/openFrameworks/libs/glfw/include -I/home/pi/openFrameworks/libs/glfw/include/GLFW -I/home/pi/openFrameworks/libs/kiss/include -I/home/pi/openFrameworks/libs/poco/include -I/home/pi/openFrameworks/libs/tess2/include -I/home/pi/openFrameworks/libs/utf8cpp/include -I/home/pi/openFrameworks/libs/utf8cpp/include/utf8 -I/home/pi/openFrameworks/libs/openFrameworks -I/home/pi/openFrameworks/libs/openFrameworks/gl -I/home/pi/openFrameworks/libs/openFrameworks/video -I/home/pi/openFrameworks/libs/openFrameworks/events -I/home/pi/openFrameworks/libs/openFrameworks/communication -I/home/pi/openFrameworks/libs/openFrameworks/math -I/home/pi/openFrameworks/libs/openFrameworks/sound -I/home/pi/openFrameworks/libs/openFrameworks/graphics -I/home/pi/openFrameworks/libs/openFrameworks/utils -I/home/pi/openFrameworks/libs/openFrameworks/types -I/home/pi/openFrameworks/libs/openFrameworks/app -I/home/pi/openFrameworks/libs/openFrameworks/3d -D_REENTRANT -pthread -I/usr/local/include/opencv -I/usr/local/include -I/usr/include/gstreamer-1.0 -I/usr/include/AL -I/usr/include/alsa -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/arm-linux-gnueabihf/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng12 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng12 -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -I./src -I/home/pi/openFrameworks/addons/ofxHttpUtils/src -I/home/pi/openFrameworks/addons/ofxOpenCv/src -I/home/pi/openFrameworks/addons/ofxOpenCv/libs -I/home/pi/openFrameworks/addons/ofxCv/libs/ofxCv/include -I/home/pi/openFrameworks/addons/ofxCv/libs/CLD/include/CLD -I/home/pi/openFrameworks/addons/ofxCv/src -Isrc -I/home/pi/openFrameworks/addons/ofxCvPiCam/src -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src/interface -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src/interface/vchiq_arm -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src/interface/mmal -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src/interface/mmal/vc -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mm [... cut ...]
 static void gray_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:199: recipe for target 'obj/linuxarmv6l/Release/src/ofApp.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/ofApp.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/devApps/RPi'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:126: recipe for target 'Release' failed
make: *** [Release] Error 2

DHaylock, 2016-01-19:
Hi @hexagon-robotics 

Check out this first https://github.com/orgicus/ofxCvPiCam/issues/3

I've just tried compiling it for Jessie on armv7l and @crx091081gb 's steps work.

Best 

David

hexagon-robotics, 2016-01-20:
Hi @DHaylock ,

Thank you for the reply but not works to me.

Best regards,
-javier

```
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h: At global scope:
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h:196:13: warning: ‘void color_callback(MMAL_PORT_T*, MMAL_BUFFER_HEADER_T*)’ declared ‘static’ but never defined [-Wunused-function]
 static void color_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h:197:13: warning: ‘void gray_callback(MMAL_PORT_T*, MMAL_BUFFER_HEADER_T*)’ declared ‘static’ but never defined [-Wunused-function]
 static void gray_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:199: recipe for target 'obj/linuxarmv7l/Release/src/ofApp.o' failed
make[1]: *** [obj/linuxarmv7l/Release/src/ofApp.o] Error 1
In file included from /home/pi/openFrameworks/apps/devApps/RPi/src/ofApp.h:14:0,
                 from /home/pi/openFrameworks/apps/devApps/RPi/src/main.cpp:2:
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h: At global scope:
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h:196:13: warning: ‘void color_callback(MMAL_PORT_T*, MMAL_BUFFER_HEADER_T*)’ declared ‘static’ but never defined [-Wunused-function]
 static void color_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h:197:13: warning: ‘void gray_callback(MMAL_PORT_T*, MMAL_BUFFER_HEADER_T*)’ declared ‘static’ but never defined [-Wunused-function]
 static void gray_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^
make[1]: Leaving directory '/home/pi/openFrameworks/apps/devApps/RPi'
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:126: recipe for target 'Release' failed
make: *** [Release] Error 2
```

```
pi@raspberrypi:~/openFrameworks/apps/devApps/RPi $ sudo ldconfig -p | grep mmal
	libmmal_vc_client.so (libc6,hard-float) => /opt/vc/lib/libmmal_vc_client.so
	libmmal_util.so (libc6,hard-float) => /opt/vc/lib/libmmal_util.so
	libmmal_core.so (libc6,hard-float) => /opt/vc/lib/libmmal_core.so
	libmmal_components.so (libc6,hard-float) => /opt/vc/lib/libmmal_components.so
	libmmal.so (libc6,hard-float) => /opt/vc/lib/libmmal.so

```

DHaylock, 2016-01-20:
Hi @hexagon-robotics,

Could you try compiling the example-ofxCvPiCam to see what happens?

David

hexagon-robotics, 2016-01-20:
I get a error with stable branch of OF.

Thanks.

```
In file included from /usr/local/include/opencv2/core/cvstd.inl.hpp:48:0,
                 from /usr/local/include/opencv2/core.hpp:3164,
                 from /usr/local/include/opencv2/opencv.hpp:46,
                 from /home/pi/openFrameworks/addons/ofxCv/src/ofxCv.h:4,
                 from /home/pi/openFrameworks/addons/ofxCv/libs/ofxCv/include/ofxCv/Kalman.h:3,
                 from ../../../addons/ofxCv/libs/ofxCv/src/Kalman.cpp:1:
/usr/include/c++/4.9/complex:390:5: note: template<class _Tp> std::complex<_Tp> std::operator*(const std::complex<_Tp>&, const _Tp&)
     operator*(const complex<_Tp>& __x, const _Tp& __y)
     ^
/usr/include/c++/4.9/complex:390:5: note:   template argument deduction/substitution failed:
../../../addons/ofxCv/libs/ofxCv/src/Kalman.cpp:32:26: note:   ‘cv::MatCommaInitializer_<float>’ is not derived from ‘const std::complex<_Tp>’
    KF.transitionMatrix = *(Mat_<T>(6, 6) <<
                          ^
In file included from /usr/local/include/opencv2/core/cvstd.inl.hpp:48:0,
                 from /usr/local/include/opencv2/core.hpp:3164,
                 from /usr/local/include/opencv2/opencv.hpp:46,
                 from /home/pi/openFrameworks/addons/ofxCv/src/ofxCv.h:4,
                 from /home/pi/openFrameworks/addons/ofxCv/libs/ofxCv/include/ofxCv/Kalman.h:3,
                 from ../../../addons/ofxCv/libs/ofxCv/src/Kalman.cpp:1:
/usr/include/c++/4.9/complex:381:5: note: template<class _Tp> std::complex<_Tp> std::operator*(const std::complex<_Tp>&, const std::complex<_Tp>&)
     operator*(const complex<_Tp>& __x, const complex<_Tp>& __y)
     ^
/usr/include/c++/4.9/complex:381:5: note:   template argument deduction/substitution failed:
../../../addons/ofxCv/libs/ofxCv/src/Kalman.cpp:32:26: note:   ‘cv::MatCommaInitializer_<float>’ is not derived from ‘const std::complex<_Tp>’
    KF.transitionMatrix = *(Mat_<T>(6, 6) <<
                          ^
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:335: recipe for target '../../../addons/obj/linuxarmv7l/Release/ofxCv/libs/ofxCv/src/Kalman.o' failed
make[1]: *** [../../../addons/obj/linuxarmv7l/Release/ofxCv/libs/ofxCv/src/Kalman.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/addons/ofxCvPiCam/example-ofxCvPiCam'
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:126: recipe for target 'Release' failed
make: *** [Release] Error 2
```

DHaylock, 2016-01-20:
Hey @hexagon-robotics is there more of the error log, i can see notes but no errors.

Cheers

closed 2016-02-08 by DHaylock

Fitoma, 2016-04-22:
Hi there,

hexagon-robotics, did you get any solution for your problem? I am getting trubble from obj/linuxarmv6l/Release/src/ofApp.o too and there is no ofApp.o in there.