WatershedArts/Footfall #9: Errors during build

cw-l, 2016-01-28:
Encountered the following errors while building the app. Please help.

makefileCommon/compile.core.mk:239: recipe for target '/usr/src/openFrameworks/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/gl/ofGLUtils.o' failed
make[2]: *** [/usr/src/openFrameworks/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/gl/ofGLUtils.o] Error 1
makefileCommon/compile.core.mk:205: recipe for target 'Release' failed
make[1]: *** [Release] Error 2
make[1]: Leaving directory '/usr/src/openFrameworks/libs/openFrameworksCompiled/project'
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:126: recipe for target 'Release' failed
make: *** [Release] Error 2

DHaylock, 2016-01-28:
Hi @cw-l, 

Hmm, thats a new one.

Have you already compiled oF for the Pi?

Best

xeryuso, 2016-01-28:
I'm having the same issue and i have compiled of already.

cw-l, 2016-01-29:
I'm getting an error for compiling oF too.

makefileCommon/compile.core.mk:239: recipe for target '/usr/src/openFrameworks/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Debug/libs/openFrameworks/gl/ofGLUtils.o' failed
make[1]: *** [/usr/src/openFrameworks/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Debug/libs/openFrameworks/gl/ofGLUtils.o] Error 1
make[1]: *** Waiting for unfinished jobs....
makefileCommon/compile.core.mk:213: recipe for target 'Debug' failed
make: *** [Debug] Error 2
there has been a problem compiling Debug OF library
please report this problem in the forums

DHaylock, 2016-01-29:
Hi @cw-l and @xeryuso 

I know these questions are like sucking eggs. 

I can see you are compiling the armv6l oF.
But ,what RPi are you compiling on 1 or 2?
and which Rasbian version (Wheezy or Jessie) are you using?

Best

cw-l, 2016-01-29:
Sorry, wasn't clear in my previous post. I'm compiling for RPi 2, on Jessie. Thanks for your help.

xeryuso, 2016-01-29:
I'm compiling for RPi 2(Armv7l), on Jessie.Thanks for your help.

DHaylock, 2016-01-29:
Hi both,

Ok originally Footfall was written for RPi 1 using Wheezy so there maybe some issues there.
I'm currently compiling for Jessie, so bare with me.

You could try doing a `make clean` in the repo the compile

cw-l, 2016-01-29:
@DHaylock Thanks, appreciate your help!

DHaylock, 2016-01-29:
Hi both,

I've managed to compile Footfall on RPi 2 using Jessie, so I will create a branch in the next couple of hours

Best

DHaylock, 2016-01-29:
Hi

RPi2 should be on here.

Best

cw-l, 2016-01-30:
Managed to compile. Thanks!

closed 2016-01-30 by cw-l