WatershedArts/Footfall #10: Error on make run

cw-l, 2016-02-01:
I'm using RPi2 running on Jessie. I can compile both OF and Footfall. Also tested that the camera works for video. Encountered the following the error.

> [notice ] ofAppEGLWindow: setupPeripherals(): native event setup complete
> [verbose] ofxCvPiCam()
> Loading Settings
> Not Loaded Settings: Program will not run!
> OpenCV Error: Assertion failed (npoints >= 0 && (depth == CV_32S || depth == CV_32F)) in approxPolyDP, file /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/contours.cpp, line 1871
> terminate called after throwing an instance of 'cv::Exception'
>   what():  /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/contours.cpp:1871: error: (-215) npoints >= 0 && (depth == CV_32S || depth == CV_32F) in function approxPolyDP
> 
> Aborted
> ../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:168: recipe for target 'run' failed
> make: *** [run] Error 134

DHaylock, 2016-02-01:
Hi @cw-l 

Ok that sounds like you dont have the xml file in the data folder

Best

cw-l, 2016-02-02:
I reinstalled OF for ARM v7 and Footfall for RPi2 and it's working now.

closed 2016-02-02 by cw-l