WatershedArts/Footfall #12: Config

AragCF, 2016-02-08:
Hello.
I am using RPi2 version of Footfall.
I need to tweak it but I don't understand the settings stored in config.xml.
Please provide an info about the meaning of each config option.

Thanks in advance.

DHaylock, 2016-02-08:
Hi @AragCF 

Ok so:

"history" : is the length of frames the MOG2 holds to adapt to the background. this is filled continuously!
"mogthreshold" : is the threshold to describe the background model
"maxdistance" : is the furthest distance a blob or person can travel in a set number of frames
"persistance" : is how long you want to track the blob before it starts being deleted
"minarea" : is the smallest a blob can be, this is the radius of the blob. Anything below this will not be recognised.
"maxarea": again is the largest a blob can be.
"contourthreshold" : is the threshold of the blob detection, so if the blob is white a value of around 200 is ok .
"threshold" : is the generic threshold of the image
"blur": blurs the incoming image
"lineypos": the point you want to read the detection from  
"linewidth": a threshold area for the above
"brightness": how much to lighten the image incase it is too dark
"maskpoints": this is where you define the mask shape

Hope it helps

AragCF, 2016-02-09:
Thanks. Trying to deal with it.

closed 2016-02-09 by DHaylock

DHaylock, 2016-02-09:
I'll add these details to the readme for future reference :)

AragCF, 2016-02-09:
Great!!!