WatershedArts/Footfall #13: Display options

AragCF, 2016-02-09:
Hello there.
Please tell me how to turn on the original image view on the screen plus the processed image.
Example: http://blogs.wcode.org/wp-content/uploads/2015/03/footfallgif.gif

Regards. Andrey.

DHaylock, 2016-02-09:
Hi @AragCF 

We took out some of the drawing stuff to save on processing, but if you are using the Pi2 it should be fine.

I'll add this back in later today.

In ofApp.cpp:Line 257 add the following in:

<pre>drawMat(frame,320,0,320,240); // Gray Scale Image
drawMat(maskOutput,0,240,320,240); // maskImage
drawMat(fgMaskMOG2,320,240,320,240); // background </pre>

They are the main images you will need.

Best

David

closed 2016-02-09 by DHaylock

AragCF, 2016-02-09:
Thanks a lot. I'll try it.

paranoidsystems, 2016-05-20:
I am trying to get this working as well. I have the following lines in apps/myapps.RPi2/src/ofApp.cpp
// draw result of output
drawMat(output, 0 0,320,240);
drawMat(frame,320,0,320,240); // Gray Scale Image
drawMat(maskOutput,0,240,320,240); // maskImage
drawMat(fgMaskMOG2,320,240,320,240); // background 

I see no changes. I assumed then I had to re "make" the app. so I copied out that file and config.xml.....deleted all the files copy the original files across copied back my 2 edited files. ran a make.......compiled fine......then started the app. but I still have only the one camera view.....

I am extremely new to PHP and Linux so perhaps I am missing a trick?  having the 3 extra views running would be handy while diagnosing my first test unit.