WatershedArts/Footfall #14: Web Interface

mo2sidefyn, 2016-02-10:
I have finally managed to compile and run the program. The people are written into the database. However, I can't display them in the webinterface. what am I doing wrong?
Many thanks for the help.

DHaylock, 2016-02-12:
Hi @mo2sidefyn,

Are there any errors in the browser?

Best

mo2sidefyn, 2016-02-12:
When i refresh the page, i can see only zeros, no peoples in or out.

With the developper tools this message shows up:

GET http://localhost/upload.php?get&interval=900 net::ERR_CONNECTION_REFUSEDsend @ jquery-1.10.2.js:8706jQuery.extend.ajax @ jquery-1.10.2.js:8136updateValues @ main-events.js:246(anonymous function) @ main-events.js:311
jquery-1.10.2.js:8706 GET http://localhost/upload.php?get&interval=900 net::ERR_CONNECTION_REFUSEDsend @ jquery-1.10.2.js:8706jQuery.extend.ajax @ jquery-1.10.2.js:8136updateValues @ main-events.js:246(anonymous function) @ main-events.js:311

DHaylock, 2016-02-15:
Hmm,

Where is your upload.php file stored?

Best

oliverhumpage, 2016-02-15:
Just to butt in here: if your dev tools log says “connection refused” when accessing http://localhost/… then it means you don’t have a web server running on your machine (at least, not on port 80).

The idea is that you have a web server somewhere that runs PHP+MySQL, where a) the pi can send its results to and b) where you can access those results. Where that is doesn’t matter so long as the pi has it set in its config, and both you and the pi can get to it. “localhost” probably isn’t that place.

Cheers,

Oliver.

mo2sidefyn, 2016-02-18:
My upload.php is in the root of my web server. The web server is on the same pi and PHP + MySQL are both running...

oliverhumpage, 2016-02-18:
> On 18 Feb 2016, at 11:28, mo2sidefyn <notifications@github.com> wrote:
> 
> My upload.php is in the root of my web server. The web server is on the same pi and PHP + MySQL are both running…

So the footfall app itself will send results to http://localhost/upload.php and that will work OK.

However, if your web browser *isn’t* on the same pi - which it likely isn’t - then connecting to “localhost” won’t work! So you need to go to http://[the.ip.address.of.your.pi]/ to see the results in your browser.

Oliver.

mo2sidefyn, 2016-02-18:
if i use the browser on my PI and go to localhost/events.php, it shows undefined.
if i navigate to localhost/upload.php the page is blank.
if i use a second computer on the same network and go to ipadressofmypi/events.php, it shows only zero's and gives me the connectionerror.

when i run the app, it counts and writes all the data into the database.

mo2sidefyn, 2016-02-18:
I changed the path of the upload.php inside of the config and main-events.js to http://ipofmypi/upload.php.
now my errorlog in the browser shows:

Uncaught TypeError: Cannot read property 'peoplein' of undefinedupdateLabels @ main-events.js:194(anonymous function) @ main-events.js:253fire @ jquery-1.10.2.js:3048self.fireWith @ jquery-1.10.2.js:3160done @ jquery-1.10.2.js:8235callback @ jquery-1.10.2.js:8778

oliverhumpage, 2016-02-19:
> if i use the browser on my PI and go to localhost/events.php, it shows undefined.

What shows “undefined”?

> if i navigate to localhost/upload.php the page is blank.

Yes - upload.php is purely for the uploading of data, it’s not meant to be viewed in a browser.

> if i use a second computer on the same network and go to ipadressofmypi/events.php, it shows only zero's and gives me the connectionerror.

What is on line 51 of main-events.js (the “url” line inside the function preLoadData())? If it’s localhost then that won’t work. Set the url to http://ipaddressofmypi/upload.php...

Oliver.

mo2sidefyn, 2016-03-03:
So i just started from scratch...
I reconfigured the whole system and rebuilt the app. On line 51 of main-events.js is http://ipaddressofmypi/upload.php

My Database is recording, but my timestamp is 0000-00-00 00:00:00 locationID 1 and event 1
After changing to %S " ofGetTimestampString("%Y-%m-%d %H:%M:%S"))" and recompiled. I've got a timecode!!
[image]
[image]

But when i navigate to http://ipofmypi/events.php...
[image]

What am i doing wrong?

DHaylock, 2016-03-03:
Hi @mo2sidefyn 

Could you try this

http://ipofmypi/upload.php?get&interval=3600

What comes back?

mo2sidefyn, 2016-03-03:
In safari nothing.
But with chrome:  Failed to load resource: the server responded with a status of 500 (Internal Server Error)

DHaylock, 2016-03-03:
Hi @mo2sidefyn 

Can you check your php log?

mo2sidefyn, 2016-03-04:
[Thu Mar 03 16:42:13.184307 2016] [:error] [pid 920] [client 000.000.00.00:00000] PHP Notice:  Undefined variable: runningtotalin in /var/www/html/upload.php on line 96

This error fills the entire log...

DHaylock, 2016-03-04:
Hi @mo2sidefyn,

In the Upload.php 
Add in $runningtotalin = 0;
[Line 73 Upload.php](https://github.com/WatershedArts/Footfall/blob/master/Web/upload.php#L79)

That should resolve that error. Strange because that definitely should be in there.

Best

mo2sidefyn, 2016-03-06:
Strange, this line is in the code...

closed 2016-04-18 by DHaylock