WatershedArts/Footfall #17: Lighting effecting the results

rrrikin, 2016-02-18:
Hello guys,

Thank you again for putting this together. I have this in a temple and have to put it up by a door. When the door opens the lights changes in the area and it miss counts. Is there anyway i can control the sensitivity or something? What should i play with in XML to get better result? currently i get results in negative as when some one leaves the area it counts more people than enters. Sometimes when i don't open the door (no light change) even then it counts reflection as object i think. Thank you for your help. This device is mounted 13 feet above the ground and its covering about six feet wide area.

Thank you 
Rikin

oliverhumpage, 2016-02-19:
You’re right that lighting plays a huge part in successful detection. Because this code essentially subtracts the current image from the background, extra light/shadows/reflections will interfere.

We spent a few days testing various locations in our entrance before finding one that worked. Even then, sometimes a shadow counts as an extra person.

In essence, without rewriting the entire image detection algorithm, you can’t put this near a door, and have to be quite careful about lighting in general.

Sorry,

Oliver.

rrrikin, 2016-02-19:
got it, thanks sadly i dont have any choice to put it somewhere else. However will see what i can do with it.

Thank you

closed 2016-02-19 by rrrikin