WatershedArts/Footfall #18: Database

MorPatates, 2016-02-18:
Hello and thanks for your previus help.I have compiled and run the program succesfully but beacuse I'm very dumb and don't know anything abouth databases and such I ran into some more problems. First a friend of mine set the database and config files according to your blog and when I tried to run the program.. It worked.. I think :d  when I pass under my camera terminal shows

200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-02-18 19:08:1455822506

lines like this but when i try to wiew them charts doesn't change. What could I be missing ?

rrrikin, 2016-02-18:
I was experiancing similar issue so it looks like you would have to change %s to %S (capital S) at two location in file ofAPP.cpp
" ofGetTimestampString("%Y-%m-%d %H:%M:%S"))"

and then it will only give you seconds in 2 digit and it will push ti MySQL

MorPatates, 2016-02-18:
Still the same but thanks for your message

bluewookie88, 2016-03-30:
This worked for me as well.

DHaylock, 2016-04-05:
Hi @MorPatates,

Could you check your browsers console?

Best

DHaylock, 2016-04-18:
@MorPatates were there any errors?

closed 2016-05-18 by DHaylock