WatershedArts/Footfall #19: Using a USB camera?

bluewookie88, 2016-02-26:
Any suggestions on where to start to try to get this stuff running with just a USB camera?  Any hints would be appreciated.

DHaylock, 2016-02-26:
Hi @bluewookie88,

If you check out the ofVideoGrabber example that should show you how to set it up.

Other than that remove the reference to the cam object: 

[Line 97](https://github.com/WatershedArts/Footfall/blob/master/RPi/src/ofApp.cpp#L97)
to Line 100. 
Replace with ofVideoGrabber code

And [Line 137](https://github.com/WatershedArts/Footfall/blob/master/RPi/src/ofApp.cpp#L137)
to 
````videoGrabber.update();
if(videoGrabber.isFrameNew()) {
       copy(videoGrabber,frame);
       // The rest of the openCv stuff
}
````

Best

David

closed 2016-02-26 by DHaylock

bluewookie88, 2016-02-27:
Thank you! Very helpful.