WatershedArts/Footfall #21: Server returning null response for fetching uploaded data

bhanuc, 2016-03-02:
I have configured everything almost correctly ( apache2 server on laptop accessible on 192.168.1.102 and all values set correctly in both config file and front-end file) Now when I run software using make run, I get logs like : 
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:18:1456949921
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:18:1456949921
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:18:1456949927
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:18:1456949927
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:18:1456949931
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:18:1456949931
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:18:1456949932
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:18:1456949938
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949940
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949946
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949949
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949951
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949956
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949960
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949961
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949965
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949972
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949972
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949976
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949976
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949980
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949982
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949985
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949985
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:19:1456949993
200:Count: 2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950001
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950001
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950002
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950010
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950010
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950019
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950021
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950022
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950025
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950029
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950031
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950035
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950038
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950045
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950047
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950052
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:20:1456950054
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950060
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950061
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950064
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950071
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950074
200:Count: 2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950077
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950081
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950084
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950085
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950089
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950092
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950096
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950101
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950101
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950106
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950110
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950114
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950116
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950118
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:21:1456950118
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950123
200:Count: 1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950124
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950124
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950133
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950134
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950134
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950137
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950137
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950138
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950140
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950143
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950146
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950148
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950156
200:Count: 2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950158
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950159
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950165
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950166
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950168
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950169
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950176
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:22:1456950177
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:23:1456950181
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:23:1456950182
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:23:1456950182
200:Count: -2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:23:1456950186
200:Count: 2</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:23:1456950188
200:Count: -1</br>Location: 1</br>Raw Timestamp: 2016-03-02 20:23:1456950192

Which seems like it is recording the data, but events page always return a json response. Where should I start debugging. Thanks

DHaylock, 2016-03-03:
Hi @bhanuc,

What json response are you getting?

bhanuc, 2016-03-03:
The response is completely empty.

DHaylock, 2016-03-03:
Is it this [ ] ?

Can you check your database to see if the data is going into it?

bhanuc, 2016-03-03:
Yes it is [ ] and in the database I just noticed that the time stamp values are all 0000-00-00 00:00:00 whereas the pi console shows the correct time stamp 2016-03-03 20:48:1457038123. rest values are correct.

DHaylock, 2016-04-05:
Hi @bhanuc,

The php script was nulling the timestamp so have removed my_real_escape_string.

You might need to change this line [Line 220] (https://github.com/WatershedArts/Footfall/blob/master/RPi2/src/ofApp.cpp#L22) to be a capital S.

closed 2016-04-18 by DHaylock