WatershedArts/Footfall #22: Realtime config tweaks

AragCF, 2016-03-08:
Hello again.

Is there a way to make the RPi app to get the config settings from some server eventually and apply this on itself (RPi)? Sorry for my english, I'll try to explain what I need.
We have an RPi2 with camera and compiled app and running with internet connection setup.
Also we have some web server that can respond with the set of config settings by special request from RPi app (Footfall), so the Footfall can get this config and apply it to itself making some tweaks available from internet (I want to change config settings from my web server to remotely manage RPi's app Footfall).

Thanks in advance.

AragCF, 2016-04-05:
Hmm?

DHaylock, 2016-04-05:
Hi @AragCF 

Sounds like you've got the setup ready. 
What response are you getting from Server JSON,XML?
You'll need to parse the data in the setup loop. 

Otherwise, this might help.
[Real time Config ofxRemoteUI](https://github.com/armadillu/ofxRemoteUI)

Best

AragCF, 2016-04-07:
I am working now on the server. It can be set up to produce both, but Json is simpler.
Do the parsed data apply in real time?
Thanks.

DHaylock, 2016-05-18:
Hi @AragCF 

They can do use ofxJSON to call the json file. But would apply them sparingly as it might confuse the system. Also it might be worth building a gui into the program to control the variables.

Best

closed 2016-05-18 by DHaylock