WatershedArts/Footfall #23: Can't get tracking right

bluewookie88, 2016-03-12:
I am having trouble adjusting the camera to count people properly.  It keeps counting one person as several.  1 = 4 or 5. I have played around adjusting things in the json file but nothing seems to fix the issue.  

Any ideas? 

[image]

raditv, 2016-03-13:
you should change minimum blob size for one, two, and three people on json file. ex: 1 person minimum 20 and two begin from 250, and three begin from 400 etc. it's depend on the size of the object on your camera.

bhanuc, 2016-04-22:
Any ideas on how to get this output on screen. I usually just ssh, but I installed a vnc server to forward the GUI to see this UI, but I am still only getting only terminal outputs. Do we need to pass a flag ?

bhanuc, 2016-04-27:
@bluewookie88 @raditv Any tips in this regard. The docs are all 404s.

oliverhumpage, 2016-04-28:
Looks like the README got updated to reference those docs, but the docs weren’t uploaded. Like it says at the top, documentation is in the process of being added at the moment. Sorry for the confusion.

The previous README is https://github.com/WatershedArts/Footfall/blob/4b298bec51b3d4f45d2442e7806cd0ccb569228a/ReadMe.md

Oliver.

DHaylock, 2016-05-18:
Hi @bluewookie88 

It depends on the height of the camera. Looking at your image people are fairly big and there doesnt seem to be much clarity in the cv image.

closed 2016-07-07 by DHaylock