WatershedArts/Footfall #25: counting cars

hardjob, 2016-03-15:
can this method be used to count cars instead of people ?

oliverhumpage, 2016-03-15:
> On 15 Mar 2016, at 04:11, hardjob <notifications@github.com> wrote:
> 
> can this method be used to count cars instead of people ?

In theory yes, but with a caveat: if you’re using a low power device like a pi, then it can only process around 10 frames per second. This means it will miss cars if you’re zoomed in too close, since they go a lot faster than people.

However, cars on a far away camera will show up fairly small, so you’ll need to tweak and test parameters to get reasonably accurate results.

In summary, I wouldn’t like to guarantee it’ll work: but if you try it out then let us know how it goes!

Oliver.

hardjob, 2016-03-18:
thanks i'll try this

closed 2016-04-05 by DHaylock