WatershedArts/Footfall #28: recipe for target 'Release' failed

nc01315, 2016-04-05:
I installed OpenFrameworks and did a make in RP2 folder on my raspberry2 and 3, I don't know the reason of fail ....

/opt/vc/include/interface/vcos/pthreads/vcos_platform.h:567:8: warning: variable ‘ok’ set but not used [-Wunused-but-set-variable]
    int ok = 0;
        ^
In file included from /usr/include/boost/system/error_code.hpp:32:0,
                 from /usr/include/boost/filesystem/path_traits.hpp:23,
                 from /usr/include/boost/filesystem/path.hpp:25,
                 from /usr/include/boost/filesystem.hpp:16,
                 from /home/pi/libs/openFrameworks/utils/ofFileUtils.h:8,
                 from /home/pi/libs/openFrameworks/ofMain.h:6,
                 from /home/pi/Downloads/Footfall/RPi2/src/main.cpp:1:
/usr/include/boost/config/abi_prefix.hpp: At global scope:
/usr/include/boost/config/abi_prefix.hpp:2:1: error: expected unqualified-id before ‘-’ token
 
 ^
In file included from /usr/include/boost/throw_exception.hpp:43:0,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:31,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/filesystem/path.hpp:29,
                 from /usr/include/boost/filesystem.hpp:16,
                 from /home/pi/libs/openFrameworks/utils/ofFileUtils.h:8,
                 from /home/pi/libs/openFrameworks/ofMain.h:6,
                 from /home/pi/Downloads/Footfall/RPi2/src/main.cpp:1:
/usr/include/boost/exception/exception.hpp:137:40: error: expected declaration before end of line
 #  pragma GCC visibility push (default)
                                        ^
/home/pi/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:200: recipe for target 'obj/linuxarmv6l/Release/src/main.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/main.o] Error 1
make[1]: Leaving directory '/home/pi/Downloads/Footfall/RPi2'
/home/pi/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2
pi@raspberrypi:~/Downloads/Footfall/RPi2 $

DHaylock, 2016-04-05:
Hi @nc01315,

From what i can see its not in the correct directory.
To compile the app you need to place it inside your openFrameworks folder.

Best

DMcGreevy, 2016-04-19:
I am getting the following errors when I make RPi2 on a Pi 3 (using Linux raspberrypi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l )

/home/pi/OF/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:199: recipe for target 'obj/linuxarmv7l/Release/src/ofApp.o' failed
make[1]: *** [obj/linuxarmv7l/Release/src/ofApp.o] Error 2
make[1]: Leaving directory '/home/pi/OF/apps/myApps/RPi2'
/home/pi/OF/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:126: recipe for target 'Release' failed
make: *** [Release] Error 2

I have changed the libs to old-libs in the ofxCvPiCam directory as was pointed out in another thread.

Anyone got any suggestions? Total noob to linux so no pointing and laughing please

Fitoma, 2016-04-19:
Same issue here. On a RPi3, OF compiled and OpenCV too, all dependencies installed, libs to old-libs and several updates/upgrades done. When making ThreadedFootFall (also RPi and RPi2 projects) I get:

`/home/pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:200: recipe for target 'obj/linuxarmv6l/Release/src/ofApp.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/ofApp.o] Error 1
make[1]: Leaving directory '/home/pi/of_v0.9.3_linuxarmv6l_release/apps/myApps/ThreadedFootfall'
/home/pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2`

infojch, 2016-04-28:
Hi,
I also happen have the same problem on my Raspberry Pi 2 (Linux raspberrypi 4.4.8-v7+ #880 SMP Fri Apr 22 21:55:04 BST 2016 armv7l) causing a failed 'Release':

```
In file included from /home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.h:14:0,
                 from /home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.cpp:7:
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h: At global scope:
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h:196:13: warning: ‘void color_callback(MMAL_PORT_T*, MMAL_BUFFER_HEADER_T*)’ declared ‘static’ but never defined [-Wunused-function]
 static void color_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h:197:13: warning: ‘void gray_callback(MMAL_PORT_T*, MMAL_BUFFER_HEADER_T*)’ declared ‘static’ but never defined [-Wunused-function]
 static void gray_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:200: recipe for target 'obj/linuxarmv6l/Release/src/ofApp.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/ofApp.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/RPi2'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
```

I have compiled OpenFrameworks 0.9.3 (v0.9.3_linuxarmv6l) prior to making RPi2. I also applied the ofxCvPiCam fix (renamed libs) without any success.

Open to any help and suggestions!
Thanks!

DMcGreevy, 2016-05-05:
@DHaylock 
I am getting the following errors when I make RPi2 on a Pi 3 (using Linux raspberrypi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l )

`/home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.cpp: In member function ‘void ofApp::makeMask()’:
/home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.cpp:60:10: error: no match for ‘operator=’ (operand types are ‘cv::Mat’ and ‘CvMat*’)
     mask = cvCreateMat(240, 320, CV_8UC1);
          ^
/home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.cpp:60:10: note: candidates are:
In file included from /usr/local/include/opencv2/core/mat.hpp:3396:0,
                 from /usr/local/include/opencv2/core.hpp:59,
                 from /usr/local/include/opencv2/opencv.hpp:46,
                 from /home/pi/openFrameworks/addons/ofxCv/src/ofxCv.h:4,
                 from /home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.h:10,
                 from /home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.cpp:7:
/usr/local/include/opencv2/core/mat.inl.hpp:560:6: note: cv::Mat& cv::Mat::operator=(const cv::Mat&)
 Mat& Mat::operator = (const Mat& m)
      ^`
`/usr/local/include/opencv2/core/mat.hpp:1102:10: note:   no known conversion for argument 1 from ‘CvMat*’ to ‘const Scalar& {aka const cv::Scalar_<double>&}’
/home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.cpp: In member function ‘void ofApp::setupCV()’:
/home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.cpp:107:70: error: invalid new-expression of abstract class type ‘cv::BackgroundSubtractorMOG2’
     pMOG2 = new BackgroundSubtractorMOG2(_history,_MOGThreshold,false);
                                                                      ^
In file included from /usr/local/include/opencv2/video.hpp:57:0,
                 from /usr/local/include/opencv2/opencv.hpp:49,
                 from /home/pi/openFrameworks/addons/ofxCv/src/ofxCv.h:4,
                 from /home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.h:10,
                 from /home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.cpp:7:
/usr/local/include/opencv2/video/background_segm.hpp:90:20: note:   because the following virtual functions are pure within ‘cv::BackgroundSubtractorMOG2’:
 class CV_EXPORTS_W BackgroundSubtractorMOG2 : public BackgroundSubtractor
                    ^
`
`/home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.cpp: In member function ‘virtual void ofApp::update()’:
/home/pi/openFrameworks/apps/myApps/RPi2/src/ofApp.cpp:158:25: error: ‘class cv::BackgroundSubtractorMOG2’ has no member named ‘operator()’
         pMOG2->operator()(maskOutput, fgMaskMOG2);
`
`/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:200: recipe for target 'obj/linuxarmv6l/Release/src/ofApp.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/ofApp.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/RPi2'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2
`

These errors are in the midst of a large number of warnings about such things as void ... declared static but never defined,  comparison between signed and unsigned integer expressions, ‘void ofLine() is deprecated  Use ofDrawLine instead,. 
Could it be that Pi3 is armv7l but openFrameworks is armv6l that is causing the problem?

Any suggestions would be much appreciated

Thanks

DHaylock, 2016-07-07:
Hi @DMcGreevy,

It could be. 

Thought RPI2 Version is armv7l.

Could you try doing a apt-get update and upgrade?

nachiarg, 2016-08-28:
Hi, I get the same message. 

> >/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:336: recipe for target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o' failed
> make[1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o] Error 1
> make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/RPi2'
> /home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
> make: *** [Release] Error

 
Raspbian

> PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
> NAME="Raspbian GNU/Linux"
> VERSION_ID="8"
> VERSION="8 (jessie)"
> ID=raspbian
> ID_LIKE=debian
> HOME_URL="http://www.raspbian.org/"
> SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
> BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
> PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)" 

openFrameworks : of_v0.9.3_linuxarmv6l_release as instructed in OF page

Alsp I did the  old libs switch

> cd ~/openFrameworks/addons/ofxCvPiCam/
> mv libs old-libs

Did anyone found a solution?

closed 2016-10-10 by DHaylock

iverdugos, 2019-06-24:
Hi Guys!

I was having the same exact problem on my Raspberry Pi 3 model B, until the guy in this link said something about the error:
 https://github.com/texane/stlink/issues/625

first 'make'. Wait for the error. Then 'make clean'. And then 'make' again. It will compile. At least it did in my Raspi... yeah!