WatershedArts/Footfall #29: Drawing contour lines in original image

davidw212, 2016-04-07:
Great app, but this seems to defeat me. The syntax is different from the opencv tutorials.

Please could you say how you drew the box and border around the identified blob in your top left image??

http://blogs.wcode.org/wp-content/uploads/2015/03/footfallgif.gif

DHaylock, 2016-05-18:
Hi @davidw212 

No problems. The syntax isnt far off.
To draw the contours. You call contourFinder.draw(); this should draw the found blobs at (0,0).

Best

closed 2016-05-18 by DHaylock

davidw212, 2016-08-19:
Sorry Ive tried for a few weeks and still cant get the video output showing in colour and with the two blob borders.

I can get the greyscale frames displayed, and a circle on the midpoint of the blob, but not an outline boundary

Please could you give an indication of which section needs amending.

thanks