WatershedArts/Footfall #30: Out Count always larger than In Count

DieterLi, 2016-04-13:
I've tested with 2 versions, RPi2 and ThreadedFootfall. Both of these versions gave me a great tally gap:

The Out Count is far larger than the In Count. I've changed the settings on blob size, because the tested cam is hanging lower than general standard, about 2.1m. 

How to adjust to get the tally gap close to zero.

Thanks in advance.

DHaylock, 2016-04-18:
Hi @DieterLi,

I'm currently working on a solution. Need to look at improving the tracking system.
However, there is always a risk that their will be false positives.

DieterLi, 2016-04-18:
Hi @DHaylock,

It's a great application for footfall counting. And in reality, it's hardly possible to get the tally, zero. But the problem I was confronted with, is the tally is too huge. 

For instance, a variance between 5% - 15% to zero, in tally, is fairly acceptable, event it's always a negative number. 

I'm still trying to adjust all the configuration parameters in order to achieve the goal. And wish you succeed on the improvement of next round. 

Great work!

closed 2016-07-07 by DHaylock