WatershedArts/Footfall #31: mysql structure

guigeek208, 2016-04-15:
Hi,
i didn't find how to create mysql database.
There is no documentation or i can't find it .
Can you help me please ?
Thanks,

oliverhumpage, 2016-04-15:
Apologies, I thought that had been uploaded.

The main dev on this is away at the moment, we’ll get the .sql file with the CREATE statements in uploaded early next week.

Oliver.

> On 15 Apr 2016, at 14:39, Guillaume Roche <notifications@github.com> wrote:
> 
> Hi,
> i didn't find how to create mysql database.
> There is no documentation or i can't find it .
> Can you help me please ?
>

DHaylock, 2016-04-18:
Hi @guigeek208 

There should be a new file added towards the end of the day.
Thought i had uploaded it previously.

Best

igordutra, 2016-04-21:
This one worked for me:

CREATE TABLE `data` (
  `timestamp` timestamp NOT NULL DEFAULT CURRENT_TIMESTAMP ON UPDATE CURRENT_TIMESTAMP,
  `locationID` tinyint(4) NOT NULL,
  `event` tinyint(4) NOT NULL
) ENGINE=InnoDB DEFAULT CHARSET=utf8 COLLATE=utf8_unicode_ci;

DHaylock, 2016-05-18:
Hi @guigeek208 

Check the new [documentation](https://github.com/WatershedArts/Footfall/blob/master/docs/server.md)

or 

@igordutra 's statement works aswell

Best

closed 2016-05-18 by DHaylock