WatershedArts/Footfall #32: Low FPS

elisiariocouto, 2016-04-19:
Hi!

I've compiled Footfall on a Raspberry Pi 2 using OpenFrameworks 0.9.3 and I can run it. Although, I'm getting really low FPS (~2) which makes it impossible to track anything.
Did anyone got this behaviour?

Thanks

DHaylock, 2016-04-21:
Hi @elisiariocouto 

How is the GPU Memory split?
Ours is set at 128mb

These details will end up in the repo soon. 

Best

elisiariocouto, 2016-04-21:
Tried with 128 and 256. No go :/

johnwestintx, 2016-05-02:
Are you using the threaded version?  I found that reducing the camera width and height in the config.json file dramatically increased my fps.  you'll have to adjust your blob sizes, but it seemed to help me.

DHaylock, 2016-05-18:
Hi @elisiariocouto 

Have you disabled the GUI?

Check your /etc/rc.local if it has startx comment it out

Might help a bit.

Best

closed 2016-07-07 by DHaylock