WatershedArts/Footfall #33: Can't compile for Rpi2

thomand, 2016-04-21:
Hi! I'm trying to use your project for a system at university, but I'm really struggeling with compiling it correctly. I've added and compiled OpenFrameworks and checked that everything is set up correctly by compiling an example and running it. That works fine. I'm getting different compilation errors (most Release 2) with footfall on a Raspberry Pi 2 model b with Jessie. I've done all the apt-get stuff and I've tried the ofxCvPiCam fix. I've naturally also added the getrepos.sh as specified in the earlier readme-file. Due to the fact that we are pressed on time with the project and are really excited (or at least was :) ) to use your code, we are really hoping that you could point us in the right direction regarding getting the Rpi2 project compiled. Thanks in advance!

DHaylock, 2016-04-21:
Hi @thomand 

Could you paste some of the errors?

Best

igordutra, 2016-04-21:
Hi, after many compilation errors I installed all add-ons manually. Then I applied the ofxCvPiCam fix and everything worked well. I'm using openFrameworks 0.9.3, Jessie and Rpi1 512Mb - it's actually running faster than I thought. 

Now, need to do some work on the web side getting some strange errors but all data is being posted on the database.

btw @DHaylock amazing project! Thanks for sharing.

thomand, 2016-04-22:
I'm going to do a reinstall of Raspbian and do everything all over just to eliminate any wrong things I've done to try to fix it. Is there anything I need to do to install each addon or can I just download each ofx-zip from github, extract it and place it in the addons-folder? The compiling is usually taking seriously long time also (1-2 hours). Do any of you have a trick up your sleeves to fix this? Thanks again for your answers! I will upload screen shots of error, if they show up again @DHaylock :)

thomand, 2016-04-22:
Hi! I got compilation problems again. Sorry for the bad pictures... :( @DHaylock and @igordutra I did a reinstall of Jessie and apt-get ++ I followed the guide on openframeworks.cc to the point (installing Armv6 although my system is armv7. That's what I did when the OF examples worked as well). I followed the tips from @igordutra and downloaded all addon zips from it's respective githubs, extracting them and adding them to the addons folder og OF. I did not do anything more with these (except mv libs old-libs in ofxCvPiCam folder). Do I have to make the addons in some way? Do they need to be added to any makefile? 
For the compiling I didn't look at the screen all the way through the compilation but I first got one error 4 (something with httputils?) and then another error 4 (for ofxCv maybe?).

Then the compilation failed on Release Error 2, like before. Is there a way to fix this? I'm probably not doing something trivial here (this is my first OF implementation), so sorry if I'm bothering you with something that's easy to fix myself. I really do hope you can help me though! :) 
Again, sorry for the bad pictures. I didn't get to do a screenshot as I was in console view...

[image]
[image]
[image]
[image]
[image]

bhanuc, 2016-04-22:
Don't use -jn with make and reduce the ram allocated to your gpu from raspi-config. It solved the same issue for me.

mhkiyani2000, 2016-04-23:
hello sir
the bellow link dont work
https://github.com/WatershedArts/Footfall/blob/master/docs/rpi.md

i cant download footfall document  in github

ericpang, 2016-04-24:
I cannot make.... can pls advise how to solve this error? Many thanks!!

[image]

thomand, 2016-04-24:
@ericpang I haven't gotten it to work myself yet. But, it seems that your application directory is not correct. You should see the example on the setup guide for Raspberry pi on openFrameworks website (http://openframeworks.cc/setup/raspberrypi/raspberry-pi-getting-started/). Your app directory should be like this "/home/pi/openFrameworks/apps/myApps/myRpiApp" (the app to levels lower than OF main folder.) This is important for the compilation.

It might also look like you did not rename the libs folder in ofxCvPiCam. Navigate to openFrameworks/addons/openCvPiCam/ in terminal and type "mv libs old-libs" (rename the folder). This makes OF use your systems MMAL instead of the one in ofxCvPiCam. You can read more about this bug here (https://github.com/orgicus/ofxCvPiCam/issues/3).

Good luck! I'm trying to compile yet again at this very moment. Hopes up! :)

thomand, 2016-04-24:
I tried just "sudo make" (i tried without sudo first but then it failed right away giving some permission stuff was missing). I still get the Release 2 error. I did not make clean first. So I tried to make clean and delete the Linuxarmv6l folder from obj. I'm trying to compile yet again. Are there something I'm doing teribly wrong here @DHaylock @igordutra ? I haven't edited any of the parameters in the code yet just to see that it gets compiled first. Is this a bad idea? I haven't gotten to setting up a server and db yet...

[image]

thomand, 2016-04-24:
I got it to compile!! :dancer: Thanks for all the help!

closed 2016-04-24 by thomand

ericpang, 2016-04-24:
Can share how you did it??

thomand, 2016-04-24:
Of course. Sorry. Installed OF for linuxarmv6 on Raspberry Pi 2 model b v1.2 compiled OF. Added all the addons for footfall manually to OF/addons (downloaded all the zips, unzipped and copied them in). Then I renamed ofxCvPiCam libs to old-libs, so that MMAL of the Raspberry is to be used. I then Copied the RPi2 folder into OF/apps/myApps directory. cd into RPi2. entered sudo make (without any -jn flags. So just "sudo make"). And I also edited the memory split first. I gave the GPU 64. The reason, i think that I got the last Release error was that I did not do the following after the compilation failed:

* in RPi2 directory do: sudo make clean (removes the build, i think)
* I noticed that linuxarmv6l folder was still there. And there was some hidden files still in it.
*In terminal i entered "sudo rm -r linuxarmv6l". This removes the entire folder and all files.
*sudo make than made it all the way through.

i also did the compilation in console view (not in the GUI if you have Raspbian gui).

Hope that might help @ericpang

mhkiyani2000, 2016-04-27:
the bellow link dont work
https://github.com/WatershedArts/Footfall/blob/master/docs/rpi.md

i cant download footfall document in github

ericpang, 2016-04-28:
Use this

https://github.com/WatershedArts/Footfall

mhkiyani2000, 2016-04-28:
hello sir
i cant khnow ,how compile
https://github.com/WatershedArts/Footfall

dont exist cmake file
could you please send me a installation document in raspberi pi2
I really appreciate your help
thank you for your time

DMcGreevy, 2016-05-05:
@mhkiyani2000 

https://recordnotfound.com/Footfall-WatershedArts-72894

mhkiyani2000, 2016-05-24:
hello
i follow the user guid but face this err in rpi2 and jessi 2016
i rplace addan\libs to old-libs too

/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/
compile.project.mk:200: recipe for target
'obj/linuxarmv6l/Release/ofxCvPiCam/example-ofxCvPiCam-calibration/src/testApp.o'
failed
make[1]: ***
[obj/linuxarmv6l/Release/ofxCvPiCam/example-ofxCvPiCam-calibration/src/testApp.o]
Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/RPi2'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/
compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2

mhkiyani2000, 2016-05-25:
hello all
i follow the user guid but face this err in rpi2 and jessi 2016
i rplace addan\libs to old-libs too

/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/
compile.project.mk:200: recipe for target
'obj/linuxarmv6l/Release/ofxCvPiCam/example-ofxCvPiCam-calibration/src/testApp.o'
failed
make[1]: ***
[obj/linuxarmv6l/Release/ofxCvPiCam/example-ofxCvPiCam-calibration/src/testApp.o]
Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/RPi2'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/
compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2

can you upload a raspberry pi2 image for this work to download??

nachiarg, 2016-08-28:
Hi, I get the same message. 

> > /home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:336: recipe for target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o' failed
> make[1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o] Error 1
> make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/RPi2'
> /home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
> make: *** [Release] Error

 
Raspbian

> PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
> NAME="Raspbian GNU/Linux"
> VERSION_ID="8"
> VERSION="8 (jessie)"
> ID=raspbian
> ID_LIKE=debian
> HOME_URL="http://www.raspbian.org/"
> SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
> BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
> PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)" 

openFrameworks : of_v0.9.3_linuxarmv6l_release as instructed in OF page

Alsp I did the  old libs switch

> cd ~/openFrameworks/addons/ofxCvPiCam/
> mv libs old-libs

Did anyone found a solution?

jagjitphull, 2017-02-08:
Hello Everyone,

       I have got the RPi2 app compiled successfully. However, my usb camera isn't being detected.

1) Is it necessary that I have to have pi camera.
2) How to configure the USB webcam to  work in lieu of pi camera.

Thanx n Regards,
Jagjit (Jp)