WatershedArts/Footfall #36: Has anyone got this project running on a Pi3

DMcGreevy, 2016-05-09:
I seem to be having all the same errors that were appearing on the Pi2 but the fixes aren't working on the Pi3. Has anyone had success and was there anything in particular different needed doing?

alisios, 2016-05-12:
Hi,

With the Rpi2 with latest Raspbian Jessie with kernel 4.4.x works to me. 

http://imagizer.imageshack.us/v2/800x600q90/921/MUDeE8.jpg

https://www.raspberrypi.org/forums/viewtopic.php?f=76&t=131830&start=25

Best regards

ttn87, 2016-05-20:
same question here , ive tried to install in rpi3 but finally not compiles.

@DHaylock  is posible to install this in pi3? thankyou

paranoidsystems, 2016-05-20:
I am running on a pi3.  https://github.com/WatershedArts/Footfall/blob/master/docs/troubleshooting.md this solved my compile failing.

ttn87, 2016-05-20:
Hi paranoidsystems please can you explain simple steps to install all and what version of OF i mean : 

- step 1
Download an compile....
Step 2...

.... because in the guide of the repository its no clear at all and i dont know if im doing right

Thankyou

paranoidsystems, 2016-05-20:
Hi ttn87,

I followed the guide in one of the older readme's here (https://github.com/WatershedArts/Footfall/commit/ec23d5169cf30cc0149ce4e9c7185d171c38a9ca ) give that a go and let me know.

ttn87, 2016-05-20:
Do you remember what version of OF?

paranoidsystems, 2016-05-20:
I believe I just used the latest version but I am not 100% sure.

DHaylock, 2016-07-07:
Hi All,

In theory it should run fine. 
As long as you download and compile the correct version of OF for which ever version of footfall you use.

David

closed 2016-07-07 by DHaylock