WatershedArts/Footfall #38: Video file playback problem

openedhardware, 2016-05-13:
Hi, David.
At first, thanks for your great work on people counting!!!

I have been working with your code for 2 months to find proper config parameters in config.xml.
I have realized that it is very hard to config them without video playback.

I have analyzed your source code of RPi2 for video file playback.
But unfortunately, I cannot see the video output on the screen when APP is running.

I am wondering if you could solve this problem.

Thanks.

mhkiyani2000, 2016-05-28:
hello
i can analyzed source code of RPI2 but cant see output on the screen,too
you find soultion for this problem??
thanks

DHaylock, 2016-07-07:
Hi @openedhardware and @mhkiyani2000,

Cheers.

On line https://github.com/WatershedArts/Footfall/blob/master/RPi2/src/ofApp.cpp#L265
you need to add the following.

`drawMat(frame, 320, 0,320,240);`

This should draw the Video/Camera texture to the Screen.

Then set the GL Window to 640,240
https://github.com/WatershedArts/Footfall/blob/master/RPi2/src/main.cpp#L7

`ofSetupOpenGL(640,240, OF_WINDOW);`

Hope it helps

David

closed 2016-07-07 by DHaylock