WatershedArts/Footfall #39: MSQL time stamps 0000-00-00 00:00:00

paranoidsystems, 2016-05-19:
I seem to be getting time stamps coming across as 0000-00-00 00:00:00

I am very very new to PHP and mysql so it could be be me. the location ID and the event are going in fine and the echoback to the terminal has the raw time correct. I am usign the new web files from a few hours ago  (myphp and mySQL is running on my qnap, everything is in a footfall folder. so not sure if perhaps I am missing a change somewhere. perhaps I need to edit something in main-events.js?

I can see var hostUrl = "./upload.php"; in there....should this be var hostUrl = ".footfall/upload.php"; ?

I have tried that and seems to make no effect that I can see. I really not sure where to start debugging.....pointers in the right direction would be helpful so I can learn what I am doing./ if this VBA i'd be ok!

atlazfi, 2016-05-19:
I had the same problem, but i corrected it by removing the timestamp from the upload.php code. 

Change 

```
$sqlq = "INSERT INTO `data` (`timestamp`,`locationID`,`event`) VALUES (:rawtimestamp,:location,:count)";
		$q = $DBH->prepare($sqlq);
		$q->execute(array(':rawtimestamp' => $rawtimestamp, ':count' => $count, ':location' => $location));

```

to

```
$sqlq = "INSERT INTO `data` (`locationID`,`event`) VALUES (:location,:count)";
		$q = $DBH->prepare($sqlq);
		$q->execute(array( ':count' => $count, ':location' => $location));
```

The software still works, because the mysql-table is set to use CURRENT_TIMESTAMP  as default for the timestamp-field.

paranoidsystems, 2016-05-19:
So your fix has the mySQL providing the time based upon the data hitting the table instead of the PI sending the time data. I will have a play. Thanks for the help atlazfi.

atlazfi, 2016-05-19:
Yes. I have no clue why they would differ. The real problem seemed to be, that the $rawtimestamp was formatted as "2015-05-19 12:01:45:142355555" (unix time at the end). You can test this yourself, by changing the timestamp-field to varchar(255) and check what is saved. For me having mysql stamp the field was the easier fix than digging any deeper.

paranoidsystems, 2016-05-19:
I will dig deeper just for the sake of learning if nothing else. But I assume you are correct. it's a sort of type mismatch. I will have a dig and report back.

DHaylock, 2016-05-19:
Hi Both,

I will update the repo now. This must have got lost when i updated the php stuff sorry about that.

Best

closed 2016-05-19 by DHaylock

mgutierrez80, 2016-06-02:
@paranoidsystems I did this and worked for me , sorry if I commit any mistake, it's my first time I post something.
BEFORE COMPILE, on file **ofApp.cpp** near line 197 : formIn.addFormField("rawtimestamp", ofGetTimestampString("%Y-%m-%d %H:%M:%s")); change %s by %S (Capital S), also near line 220 

On your Mysql, the type for "date" field would be : datetime 

Not need to change php, I hope this helps you.

paranoidsystems, 2016-06-02:
Thats a much better fix than my commit. I will change my system the same way. Thanks @mgutierrez80