WatershedArts/Footfall #42: Footfall RPI2 jessei Image for download

mhkiyani2000, 2016-05-25:
hello all
i follow the user guid but face this err in rpi2 and jessi 2016
i rplace addan\libs to old-libs too

/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:200: recipe for target 'obj/linuxarmv6l/Release/ofxCvPiCam/example-ofxCvPiCam-calibration/src/testApp.o' failed
make[1]: *** [obj/linuxarmv6l/Release/ofxCvPiCam/example-ofxCvPiCam-calibration/src/testApp.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/RPi2'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2

can you upload a raspberry pi2 image for this work to download??

mhkiyani2000, 2016-05-25:
got it to compile
Thanks for all
Good luck!

Fitoma, 2016-05-31:
@mhkiyani2000 Could you upload that image now you got it working? :)

evilchippy, 2016-06-11:
Here's how I installed it. Note that you can only see the output video via a monitor connected to the HDMI output. You can't see the output AT ALL with a headless setup.
 
1.	Preparing Raspberry Pi.
1)	INSTALL RASPBIAN "JESSIE".
-	Install the latest Raspbian (Debian Jessie) image from the Raspberry Pi downloads page
-	Follow the instructions at Raspberry Pi Preparing Your SD Card to get Raspbian installed onto your SD Card.
2)	CONFIGURE THE RASPBERRY PI.
-	Expand file system, enable picamera, split memory.
Type in the terminal sudo raspi-config
We need to make sure the CPU has 192MB of RAM in order to compile openFrameworks. Once you have compiled openFrameworks you may want to repeat this step with 128.
•	Select 1 Expand Filesystem and hit Enter.
•	Select Enable Camera and hit Enter.
•	Select 8 Advanced Options and hit Enter.
o	Select A3 Memory Split and hit Enter.
o	Type 64 and Hit <ok>
openFrameworks does not use the X11 Desktop for running OF applications. You can save GPU memory by disabling it. This is optional but suggested.
•	Select 3 Boot Options
•	Select B1 Console or B2 Console Autologin
Unlike Wheezy, Debian Jessie does not display the IP address on boot. You may wish to get the current IP by typing ifconfig. The IP address is under the eth0 entry for wired ethernet. This can possibly change on reboot.
•	Reboot when prompted or type sudo reboot
•	Assuming you have internet access run these commands to update the software to the latest packages.
sudo apt-get clean
sudo apt-get update
sudo apt-get upgrade
Note: The above steps may take a little while.
2.	Installing openFrameworks on the Raspberry Pi.

1)	DOWNLOAD OPENFRAMEWORKS.
You now can download openFrameworks and uncompress it into a folder. Using a Shell, the following commands will download openFrameworks and uncompress it into the folder /home/pi/openFrameworks.
wget http://openframeworks.cc/versisuons/v0.9.3/of_v0.9.3_linuxarmv6l_release.tar.gz
mkdir openFrameworks
tar vxfz of_v0.9.3_linuxarmv6l_release.tar.gz -C openFrameworks --strip-components 1
(not 1, but L)
2)	INSTALL PACKAGES AND COMPILE OPENFRAMEWORKS:
Make sure you didn't skip the Memory Split step in the above section Configure the Raspberry Pi or it will eventually fail.
The time for these steps will depend on whether you are on RPI2 and the speed of the Raspbian mirrors to download the packages. Assuming openFrameworks is located at /home/pi/openFrameworks run the following commands to install the necessary packages and compile openFrameworks.
cd /home/pi/openFrameworks/scripts/linux/debian
sudo chmod +x install_dependencies.sh
sudo ./install_dependencies.sh
You are now ready to compile openFrameworks.
make Release -C /home/pi/openFrameworks/libs/openFrameworksCompiled/project
3.	Installing People Counter Application on the Rapberry Pi
1)	INSTALL DEPENDENCIES FOR APP.
-	Copy the getrepos.sh file to the addons folder of openFrameworks. (You can copy this file with WinSCP program)
Run WinSCP and input IP address, port number (22), username (pi), password (raspberry) and connect to your RPi.
Select /home/pi/openFrameworks/addons folder.
Drag the shell script file above and drop it to the addons folder.
-	Run shell script above to install dependencies for our APP.
cd /home/pi/openFrameworks/addons
sudo chmod +x getrepos.sh
./getrepos.sh
-	Modify ofxCviCam library.
MMAL Libraries have changed slightly so users will have run into compiler errors.
cd /home/pi/openFrameworks/addons/ofxCvPiCam
mv libs old-libs

2)	BUILD PEOPLE COUNTER APP
-	Copy the source folder to the RPi and compile. (Folder name is people_counter_RPi2)
cd people_counter_RPi2
sudo make clean
sudo make 
After compile is finished, the executable file is generated in bin folder. 
-	Configure parameters for APP.
Configuration parameters are in bin/data/config.xml
Open this file and change parameters as you like.
-	Run App.
cd bin
./people_counter_RPi2
(NOTE : the APP outputs recoding videos to the HDMI port default. So you cannot see display when you are running on SSH.)

Fitoma, 2016-06-15:
Hi there,

@evilchippy you make no reference to OpenCV. Didn't you install it? I assumed it was needed for the app to work O_o

evilchippy, 2016-06-15:
Hi @Fitoma,

When you copy getrepos.sh to the addons folder of openFrameworks and run the shell script, all of the necessary dependencies are automatically downloaded. Just look at the "1) INSTALL DEPENDENCIES FOR APP" section in my previous posting. 

I can confirm that I've 2x installed this people counter with the instructions I posted without fail.

closed 2016-07-07 by DHaylock

pigsfoot, 2016-09-10:
@evilchippy have you managed to get this working on an RPI3.. followed your instructions and the main build notes and its just not working :-(

evilchippy, 2016-09-10:
I used an RPi2, not RPi3.