WatershedArts/Footfall #44: Using shapes and non-vertical lines as trigger areas

igordutra, 2016-06-28:
Hi @DHaylock 

We currently have a horizontal line to trigger traffic in and traffic out events. We're trying to position the cameras in different angles and locations, therefore the horizontal line isn't suitable for all our scenarios. I would like to make the following adjustments to the code but since I'm not familiar with openFrameworks I don't know where to start:

1. Would be possible to use vertical lines in case the camera is positioned differently?
2. Would be possible to create shapes e.g. a square so we can use it do track whether people come in and out that area? (is it done via mask parameters?)

Can you send some reference and tips on how to implement the above items?

Thanks
Igor

DHaylock, 2016-07-07:
Hi @igordutra,

Look at the ofRectangle documentation.

If you assign a ofRectangle some coordinates and a size, you can check whether the blob has gone Inside the box using this https://github.com/openframeworks/openFrameworks/blob/master/libs/openFrameworks/types/ofRectangle.h#L462

You can do multiples as well.

In terms of circles/triangles and polygons I'm sure there are ways.

David

closed 2016-07-07 by DHaylock

igordutra, 2016-07-07:
Thanks, I'll give it a go!