WatershedArts/Footfall #49: Threaded Footfall Compilation failed on Raspberry Pi

umang-sh, 2016-09-03:
I complied openframeworks as per the instructions here 
http://openframeworks.cc/setup/raspberrypi/raspberry-pi-getting-started/
Then build and compiled with footfall using this 
(https://github.com/WatershedArts/Footfall/blob/master/docs/rpi.md)
Attaching the last few lines of terminal output from raspberry Pi 3 :

/home/pi/openFrameworks/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:218:15: warning: unused variable 'principalPointYLt' [-Wunused-variable]
float principalPointYLt = xml.getFloatValue("stCamera:PerspectiveModel/stCamera:ImageYCenter");
^
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:336: recipe for target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o' failed
make[1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/ThreadedFootfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2

famaia, 2016-09-04:
Hello,
I am trying to compile this all weekend, without success, with 0.8 and 0.9 versions of OF.
My error is the same as you.
Tried with the Threaded and the RPi2 version.

Any suggestions?

DHaylock, 2017-02-18:
Hi Both, 

New readmes in repo aswell as new version of Footfall

The errors are to do with which branch of ofxCv is being compiled. See ReadMe update 3.

Best

closed 2017-02-18 by DHaylock