WatershedArts/Footfall #50: RPI3 / Jessie / Armv6 / OpenFrameworks 0.9.3 ???  Issue

pigsfoot, 2016-09-10:
Hi, 
I have read pretty much everything on here including all the closed tickets but has anyone managed to get this working on an RPI3 running Jessie. I appreciate @DHaylock says it should but no one seems to have managed it... yet

I keep getting the same error everyone else seems to get when compiling the Source Code which seems to point to a missing file "/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o"

`
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:336: recipe for target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o' failed
make[1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o] Error 2
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/RPi2'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2
`

Not sure what should create this file, is it an OpenFrameWorks problem, Jessie, RPI3 or to do with this application?

Unfortunately RPI2's are becoming very hard to get so really need to get this working on an RPI3. I haven't got an RPi2 to try either :-(

@evilchippy - You seem to have got it working several time reading this post - [(https://github.com/WatershedArts/Footfall/issues/42)](url)
Any such luck on an RPI3

This is the OpenFrameworks install instructions i am following, still based on Armv6 as the V7 build dosn't seem to be supported on an RPI3 - [http://openframeworks.cc/setup/raspberrypi/raspberry-pi-getting-started/](url)

I have ran apt-get update and upgrade several times, also renamed the libs folder to old-libs

Does anyone have a working .img file they could share? To show my appreciation of providing a working build on an RPI3 I'm happy to make a contribution to a charity of their choice. I've spent close to 9 hours so far and several new builds trying to get this working :-)

Thanks.

evilchippy, 2016-09-10:
> WatershedArts/Footfall/issues/42) <http://url>

Hi There,

Here's my experience with the people counter:

I spent countless hours trying to get this to work in a production
environment. I used an RPi2, and the install is not the difficult part.

Tuning the people counter to count accurately in a typical production
environment is wildly time consuming. Even using video playback as an
imput, you can easily spend 30 hours fussing with the settings. The problem
stems from the ambient light level changing. Add shadows to the mix, and
you're on a world of hurt.

As it stands now, this was not a cost effective for me to count people.
Computer vision is in it's own category of computer science.

pigsfoot, 2016-09-10:
@evilchippy have you found an alternative solution, I'm keen to find a people tracker / counter system i can use in my shop and also outside the door to count how many go past and how many come in..

evilchippy, 2016-09-11:
http://www.trafsys.com/people-counting/

titusece, 2016-09-13:
@pigsfoot I've got the same error like you.
https://github.com/WatershedArts/Footfall/issues/51
Any news about this issue ?
Can you please help me on this problem ?

In file included from /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:1:0:
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/include/ofxCv/Calibration.h: In constructor ‘ofxCv::Calibration::Calibration()’:
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/include/ofxCv/Calibration.h:116:9: warning: ‘ofxCv::Calibration::reprojectionError’ will be initialized after [-Wreorder]
   float reprojectionError;
         ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/include/ofxCv/Calibration.h:111:11: warning:   ‘cv::Mat ofxCv::Calibration::distCoeffs’ [-Wreorder]
   cv::Mat distCoeffs;
           ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:98:5: warning:   when initialized here [-Wreorder]
     Calibration::Calibration() :
     ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp: In member function ‘void ofxCv::Calibration::loadLcp(std::string, float, bool)’:
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:163:9: _**error: ‘ofXml’ was not declared in this scope
         ofXml xml;**_
         ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:164:23: _**error: ‘xml’ was not declared in this scope**_
         bool loaded = xml.load(ofToDataPath(filename, absolute));
                       ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:233:19: warning: unused variable ‘principalPointXGt’ [-Wunused-variable]
             float principalPointXGt = xml.getFloatValue("stCamera:PerspectiveModel/stCamera:ImageXCenter") ;
                   ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:234:19: warning: unused variable ‘principalPointYGt’ [-Wunused-variable]
             float principalPointYGt = xml.getFloatValue("stCamera:PerspectiveModel/stCamera:ImageYCenter");
                   ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:207:15: warning: unused variable ‘principalPointX’ [-Wunused-variable]
         float principalPointX = 0.5; // ImageXCenter, ratio
               ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:208:15: warning: unused variable ‘principalPointY’ [-Wunused-variable]
         float principalPointY = 0.5; // ImageYCenter, ratio
               ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:220:15: warning: unused variable ‘principalPointXLt’ [-Wunused-variable]
         float principalPointXLt = xml.getFloatValue("stCamera:PerspectiveModel/stCamera:ImageXCenter");
               ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:221:15: warning: unused variable ‘principalPointYLt’ [-Wunused-variable]
         float principalPointYLt = xml.getFloatValue("stCamera:PerspectiveModel/stCamera:ImageYCenter");
               ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:336: recipe for target '/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o' failed
make[1]: *** [/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o] Error 1
make[1]: Leaving directory '/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/apps/myApps/people_counter_RPi2'
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2

Regards,
Titus S.

AdarshMV, 2017-01-30:
Hey! I was able to get it up and running on the Raspberry Pi 3. Not sure if you faced the same issues I did, but this blog helped: 
[http://www.oodlestechnologies.com/blogs/People-count-by-raspberry-pi-3-using-openframework]
Following it word to word got it up and running. Just make sure you start with a fresh Jessie image (some openCV libraries I had installed on my earlier image conflicted with the build)

Hope this helps!

adarshmammen, 2017-01-30:
Hey! I was able to get it up and running on the Raspberry Pi 3. Not sure if you faced the same issues I did, but this blog helped:
http://www.oodlestechnologies.com/blogs/People-count-by-raspberry-pi-3-using-openframework
Following it word to word got it up and running. Just make sure you start with a fresh Jessie image (some openCV libraries I had installed on my earlier image conflicted with the build)

Hope this helps!

titusece, 2017-01-31:
Sounds good.
Thanks.

closed 2017-02-18 by DHaylock