WatershedArts/Footfall #51: People counter in RPi3.

titusece, 2016-09-13:
Dear All,
I'm able to build the openframeworks linuxarmv6l.

**Here is the log:**
Creating library  /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a
ar -cr "/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a" /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/events/ofEvents.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/types/ofColor.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/types/ofParameter.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/types/ofRectangle.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/types/ofParameterGroup.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/types/ofBaseTypes.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/app/ofMainLoop.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/app/ofAppNoWindow.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/app/ofAppRunner.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/app/ofAppEGLWindow.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/graphics/ofCairoRenderer.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/graphics/ofTessellator.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/graphics/ofRendererCollection.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/graphics/ofPixels.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/graphics/ofPath.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/graphics/ofBitmapFont.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/graphics/ofImage.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/graphics/ofTrueTypeFont.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/graphics/of3dGraphics.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/graphics/ofPolyline.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/graphics/ofGraphics.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/math/ofVec4f.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/math/ofMatrix4x4.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/math/ofMatrix3x3.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/math/ofQuaternion.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/math/ofVec2f.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/math/ofMath.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/communication/ofSerial.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/communication/ofArduino.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/3d/ofNode.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/3d/of3dPrimitives.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/3d/of3dUtils.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/3d/ofEasyCam.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/3d/ofMesh.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/3d/ofCamera.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/video/ofVideoPlayer.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/video/ofVideoGrabber.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/video/ofGstUtils.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/video/ofGstVideoPlayer.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/video/ofGstVideoGrabber.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/sound/ofSoundBuffer.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/sound/ofFmodSoundPlayer.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/sound/ofSoundStream.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/sound/ofOpenALSoundPlayer.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/sound/ofBaseSoundStream.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/sound/ofSoundPlayer.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/sound/ofRtAudioSoundStream.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/gl/ofGLRenderer.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/gl/ofVboMesh.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/gl/ofTexture.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/obj/Release/libs/openFrameworks/gl/ofShader.o /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/lib/linuxarmv6l/ob [... cut ...]
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:142: recipe for target 'Debug' failed
make[1]: *** [Debug] Error 2
make[1]: Leaving directory '/home/pi/workdir/rPi/camera_pi/people_counter'
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:165: recipe for target 'all' failed
make: *** [all] Error 2
pi@raspberrypi:~/workdir/rPi/camera_pi/people_counter $
pi@raspberrypi:~/workdir/rPi/camera_pi/people_counter $

Can anyone help me on this issue and what could be the problem ?

Regards,
Titus S.

titusece, 2016-09-13:
Hello All,

This is the current build error log for people counter app.

In file included from /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/src/ofxCv.h:23:0,
                 from /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/include/ofxCv/Calibration.h:22,
                 from /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:1:
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/include/ofxCv/Tracker.h: In member function ‘virtual const std::vector<unsigned int>& ofxCv::RectTracker::track(const std::vector<cv::Rect_<int> >&)’:
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/include/ofxCv/Tracker.h:324:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    for(int i = 0; i < labels.size(); i++) {
                     ^
In file included from /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:1:0:
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/include/ofxCv/Calibration.h: In constructor ‘ofxCv::Calibration::Calibration()’:
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/include/ofxCv/Calibration.h:116:9: warning: ‘ofxCv::Calibration::reprojectionError’ will be initialized after [-Wreorder]
   float reprojectionError;
         ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/include/ofxCv/Calibration.h:111:11: warning:   ‘cv::Mat ofxCv::Calibration::distCoeffs’ [-Wreorder]
   cv::Mat distCoeffs;
           ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:98:5: warning:   when initialized here [-Wreorder]
     Calibration::Calibration() :
     ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp: In member function ‘void ofxCv::Calibration::loadLcp(std::string, float, bool)’:
_/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:161:9: **error: ‘ofXml’ was not declared in this scope**
         ofXml xml;
         ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:162:23: **error: ‘xml’ was not declared in this scope**
         bool loaded = xml.load(ofToDataPath(filename, absolute));_
                       ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:230:19: warning: unused variable ‘principalPointXGt’ [-Wunused-variable]
             float principalPointXGt = xml.getFloatValue("stCamera:PerspectiveModel/stCamera:ImageXCenter") ;
                   ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:231:19: warning: unused variable ‘principalPointYGt’ [-Wunused-variable]
             float principalPointYGt = xml.getFloatValue("stCamera:PerspectiveModel/stCamera:ImageYCenter");
                   ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:204:15: warning: unused variable ‘principalPointX’ [-Wunused-variable]
         float principalPointX = 0.5; // ImageXCenter, ratio
               ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:205:15: warning: unused variable ‘principalPointY’ [-Wunused-variable]
         float principalPointY = 0.5; // ImageYCenter, ratio
               ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:217:15: warning: unused variable ‘principalPointXLt’ [-Wunused-variable]
         float principalPointXLt = xml.getFloatValue("stCamera:PerspectiveModel/stCamera:ImageXCenter");
               ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:218:15: warning: unused variable ‘principalPointYLt’ [-Wunused-variable]
         float principalPointYLt = xml.getFloatValue("stCamera:PerspectiveModel/stCamera:ImageYCenter");
               ^
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:336: recipe for target '/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o' failed
make[1]: *** [/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/obj/linuxarmv6l/Release/ofxCv/libs/ofxCv/src/Calibration.o] Error 1
make[1]: Leaving directory '/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/apps/myApps/people_counter_RPi2'
/home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2
pi@raspberrypi:~/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/apps/myApps/people_counter_RPi2 $ vim /home/pi/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/src/Calibration.cpp
pi@raspberrypi:~/workdir/rPi/camera_pi/of_v0.9.3_linuxarmv6l_release/apps/myApps/people_counter_RPi2 $

Can any one suggest clue on this issue ?

Regards,
Titus S.

adarshmammen, 2017-01-30:
Hey! I was able to get it up and running on the Raspberry Pi 3. Not sure if you faced the same issues I did, but this blog helped:
http://www.oodlestechnologies.com/blogs/People-count-by-raspberry-pi-3-using-openframework
Following it word to word got it up and running. Just make sure you start with a fresh Jessie image (some openCV libraries I had installed on my earlier image conflicted with the build)

Hope this helps!

crertel, 2017-02-10:
@adarshmammen great work!

Have you noticed if the Pi gets particularly hot when running this code?

I'd like to use this for an embedded application, but I'm trying to see how much I need to worry about thermal issues.

adarshmammen, 2017-02-10:
@crertel The Pi can handle it, but it's only putting out an average of 3 fps, I'm still not sure if I can up this by tweaking the parameters or whether that's all the Pi can churn out.

closed 2017-02-18 by DHaylock