WatershedArts/Footfall #54: rotate the camera view

0raid, 2016-11-25:
Hello - nice project!

one quick question , is it possible to rotate the camera view ???

i have just setup this on my store , and i have install the camera the other-way , so  going in (is going out) and the opposite.

thanks

DHaylock, 2016-11-25:
Hi @0raid 

You should be able to flip the camera view using this line.

[https://github.com/WatershedArts/Footfall/blob/master/RPi/src/ofApp.cpp#L98](https://github.com/WatershedArts/Footfall/blob/master/RPi/src/ofApp.cpp#L98)

Hope this helps

David

0raid, 2016-11-25:
hello DHaylock!

thanks for you fast reply! it works pefect! thanks again

one more question is it possible to change the position of the counting line???? 

thanks again!

george

DHaylock, 2016-11-25:
Hi @0raid 

Yeah so if you alter this config

[https://github.com/WatershedArts/Footfall/blob/master/RPi/bin/data/config.xml#L14](https://github.com/WatershedArts/Footfall/blob/master/RPi/bin/data/config.xml#L14)

Should work 

David

closed 2017-02-13 by DHaylock