WatershedArts/Footfall #55: Debugging display output

paranoidsystems, 2016-12-09:
Hi All,

My test pi3 is in place and counting people. I really need to change blob size and direction etc. but I only have access to the unit with ssh or VNC now so no HDMI to see the display. I understand over VNC you can't see the camera display because of the way the pi dumps this right to the video buffer. Is it possible to be able to stream or record this output? Ideally it would be nice to go to http://mypi.com:8080 or something and see the "display". I am not sure how else to debug my strange counts without seeing the output.

DHaylock, 2016-12-09:
Hi @paranoidsystems,

We also had this issue on numerous deployments of Footfall. 
We cheated slightly by saving images directly from the screen every second while there was a blob in the camera view. That should give you a small sequence to play with.

You'll need to add this in ofSaveFrame(false);

Hope this helps

paranoidsystems, 2016-12-09:
That should do :) will try it out in a bit.

closed 2017-02-13 by DHaylock