WatershedArts/Footfall #59: White screen of death on the new Footfall RPi2 of_v0.9.8_linuxarmv6l

d000000h, 2017-02-21:
Ive succesfully compiled of_v0.9.8_linuxarmv6l_release and Footfall, i had to do the libs to old-libs trick and use the VS2012-static branch of OfxCv to get it all to compile, i did have Threaded Footfall running before this new version.

Im using RASPBERRY PI CAMERA BOARD V2

Now when i startup Footfall i get just a white screen;

`pi@raspberrypi:~/openFrameworks/apps/myApps/Footfall/bin $ ./Footfall
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): screenRect: 1920x1080
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): windowRect: 1024x768
[notice ] ofAppEGLWindow: createSurface(): setting up EGL Display
[notice ] ofAppEGLWindow: createSurface(): EGL Display correctly set 0x1
[notice ] ofAppEGLWindow: createSurface(): no current renderer selected
[notice ] ofAppEGLWindow: createSurface(): default renderer detected
[notice ] ofAppEGLWindow: createSurface(): surface created correctly
[notice ] ofAppEGLWindow: createSurface(): API bound correctly
[notice ] ofAppEGLWindow: createSurface(): -----EGL-----
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MAJOR = 1
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MINOR = 4
[notice ] ofAppEGLWindow: createSurface(): EGL_CLIENT_APIS = OpenGL_ES OpenVG
[notice ] ofAppEGLWindow: createSurface(): EGL_VENDOR = Broadcom
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION = 1.4
[notice ] ofAppEGLWindow: createSurface(): EGL_EXTENSIONS = EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_vg_parent_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_lock_surface
[notice ] ofAppEGLWindow: createSurface(): GL_RENDERER = VideoCore IV HW
[notice ] ofAppEGLWindow: createSurface(): GL_VERSION  = OpenGL ES-CM 1.1
[notice ] ofAppEGLWindow: createSurface(): GL_VENDOR   = Broadcom
[notice ] ofAppEGLWindow: createSurface(): -------------
[notice ] ofAppEGLWindow: setupPeripherals(): peripheral setup complete
[notice ] ofAppEGLWindow: setupNativeUDev(): created udev object
[notice ] ofAppEGLWindow: setupMouse(): unabled to find mouse
[ error ] ofAppEGLWindow: setupMouse(): did not open mouse
[ error ] ofAppEGLWindow: setupMouse(): did not open mouse, mouse_fd < 0
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard_fd= 15 devicePath=/dev/input/by-path/platform-3f980000.usb-usb-0:1.3:1.0-event-kbd
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard device name =   USB Keyboard
[notice ] ofAppEGLWindow: setupPeripherals(): native event setup complete
`

This is what i see on the screen:
[image]

What is causing this; why can i run the old version but not the new one, please help me

DHaylock, 2017-02-21:
Hi @d000000h,

Hmm.

What is in you config.json?

DHaylock, 2017-02-21:
could you try running one of the ofxCvPiCam examples?

d000000h, 2017-02-21:
Wow you are so fast :)

My config is completely stock 

`{
        "Footfall": {
                "AppConfig": {
                        "usehttp": false,
                        "usecsvlogging": false
                },
                "CameraConfig": {
                        "camerawidth":160,
                        "cameraheight":120,
                        "fliphorizontally": false,
                        "flipvertically": false,
                        "trackshadows": true,
                        "threshold": 200,
                        "mogthreshold": 16,
                        "shadowpixelratio": 0.25,
                        "dilate": 5,
                        "erode": 3,
                        "history": 100,
                        "blur": 5,
                        "usemask": false,
                        "showshadowimage": true,
                        "MaskArea":
                        [
                         {"coordx":0,"coordy":0},
                         {"coordx":160,"coordy":0},
                         {"coordx":160,"coordy":120},
                         {"coordx":0,"coordy":120}
                         ]
                },
                "TrackingConfig": {
                        "threshold":10,
                        "minarea":10,
                        "maxarea":90,
                        "flipvertically":false,
                        "startposx":0,
                        "startposy":60,
                        "offset":20,
                        "blobdyingtime": 0.50,
                        "persistance":1,
                        "maxdistance":100,
                        "minsizeone": 10,
                        "minsizetwo": 50,
                        "minsizethree": 100
                },
                "HttpConfig": {
                        "keepbackups" : false,
                        "postserver":"",
                        "postextension":"",
                        "secretkey":"",
                        "maxretries":1
                }
        }
}
`

d000000h, 2017-02-21:
I compiled one of the examples and ran it , it did give me an image on the screen and worked fine

`pi@raspberrypi:~/openFrameworks/addons/ofxCvPiCam/example-ofxCvPiCamSimple/bin $ ./example-ofxCvPiCamSimple
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): screenRect: 1920x1080
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): windowRect: 1280x720
[notice ] ofAppEGLWindow: createSurface(): setting up EGL Display
[notice ] ofAppEGLWindow: createSurface(): EGL Display correctly set 0x1
[notice ] ofAppEGLWindow: createSurface(): no current renderer selected
[notice ] ofAppEGLWindow: createSurface(): default renderer detected
[notice ] ofAppEGLWindow: createSurface(): surface created correctly
[notice ] ofAppEGLWindow: createSurface(): API bound correctly
[notice ] ofAppEGLWindow: createSurface(): -----EGL-----
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MAJOR = 1
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MINOR = 4
[notice ] ofAppEGLWindow: createSurface(): EGL_CLIENT_APIS = OpenGL_ES OpenVG
[notice ] ofAppEGLWindow: createSurface(): EGL_VENDOR = Broadcom
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION = 1.4
[notice ] ofAppEGLWindow: createSurface(): EGL_EXTENSIONS = EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_vg_parent_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_lock_surface
[notice ] ofAppEGLWindow: createSurface(): GL_RENDERER = VideoCore IV HW
[notice ] ofAppEGLWindow: createSurface(): GL_VERSION  = OpenGL ES-CM 1.1
[notice ] ofAppEGLWindow: createSurface(): GL_VENDOR   = Broadcom
[notice ] ofAppEGLWindow: createSurface(): -------------
[notice ] ofAppEGLWindow: setupPeripherals(): peripheral setup complete
[notice ] ofAppEGLWindow: setupNativeUDev(): created udev object
[notice ] ofAppEGLWindow: setupMouse(): unabled to find mouse
[ error ] ofAppEGLWindow: setupMouse(): did not open mouse
[ error ] ofAppEGLWindow: setupMouse(): did not open mouse, mouse_fd < 0
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard_fd= 15 devicePath=/dev/input/by-path/platform-3f980000.usb-usb-0:1.3:1.0-event-kbd
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard device name =   USB Keyboard
[notice ] ofAppEGLWindow: setupPeripherals(): native event setup complete
[verbose] ofxCvPiCam()
[warning] ofThread: - name: Thread 2 - Calling startThread with verbose is deprecated.
[verbose] Attached gray callback

[verbose] Capture started
`

DHaylock, 2017-02-21:
How odd ... Try cleaning the app and re building it 

<code>make clean</code>

DHaylock, 2017-02-21:
This has never happened us, the fact there is no content on the screen is odd

d000000h, 2017-02-21:
i will try again

d000000h, 2017-02-21:
[image]
[image]

Same result, still white screen

DHaylock, 2017-02-21:
That's really odd, ok try copying one of the ofxcvpicam examples to the app folder. Recompile. See if that works, if it does copy the footfall source code across to the src folder. I'll have to check this later on. Don't think it's memory based as the example works.

d000000h, 2017-02-21:
so i guess you want me to move an example to the myapps folder and see if that works from there? im trying that now ... and if that works i can move the stuff from Footfall/src/ to the example /src/ folder and recompile and it should work ? 📦 

im gonna try that

d000000h, 2017-02-21:
Compiling example in the myApps folder works fine

Moving Footfall/src to Example/src and then recompiling, 

White screen of death

PRITIAEC10, 2017-02-22:
are you using a raspberry pi 3?

d000000h, 2017-02-22:
No im using a raspberry pi 2, with armv6 openframeworks on Raspbian default image.

I even tried yesterday to make another sd card, reinstlal the raspbian, recompiled openframeworks, recompiled ofxCvPiCam example -> Works fine, Recompiled Footfall -> White screen of death

Did all the things suggested here, moved an example to the myapps folder -> compiled and worked fine -> copied Footfall/src/ to the example/src/ folder recompiled -> BAM white screen of death :(

Default Config file, 0 changes to that

PRITIAEC10, 2017-02-22:
I am using a raspberrypi 3 , with arm6 openframeworks , compiled the footfall as mentioned in the steps and when i tried to run the exe, i got the white screen of death, and i have to restart the pi, i too have compiled the RPi2 and threadedfootfall  and it was working. Is the issue related to config file, because i am using the config file as provided without any change

d000000h, 2017-02-22:
So i guess this is confirmed to bug on the RPi3 too, after rebooting it you still get the white screen ? or does it work then ?

DHaylock, 2017-02-22:
Hi @d000000h, this sounds insane but could you check the ofApp.cpp file to see if it has any content in it?

DHaylock, 2017-02-22:
the fact you dont get anything on screen is bizarre you should at least get some debug text

PRITIAEC10, 2017-02-22:
ofApp.cpp is not empty, Can it be a X11 desktop issue, 

this is i get when i control C, the white screen
pi@raspberrypi:~ $ ./startFootfall.sh 
[warning] ofAppEGLWindow: init(): X11 not availble on RPI yet, using a native window instead
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): screenRect: 1024x768
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): windowRect: 1024x768
[notice ] ofAppEGLWindow: createSurface(): setting up EGL Display
[notice ] ofAppEGLWindow: createSurface(): EGL Display correctly set 0x1
[notice ] ofAppEGLWindow: createSurface(): no current renderer selected
[notice ] ofAppEGLWindow: createSurface(): default renderer detected
[notice ] ofAppEGLWindow: createSurface(): surface created correctly
[notice ] ofAppEGLWindow: createSurface(): API bound correctly
[notice ] ofAppEGLWindow: createSurface(): -----EGL-----
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MAJOR = 1
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MINOR = 4
[notice ] ofAppEGLWindow: createSurface(): EGL_CLIENT_APIS = OpenGL_ES OpenVG
[notice ] ofAppEGLWindow: createSurface(): EGL_VENDOR = Broadcom
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION = 1.4
[notice ] ofAppEGLWindow: createSurface(): EGL_EXTENSIONS = EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_vg_parent_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_lock_surface 
[notice ] ofAppEGLWindow: createSurface(): GL_RENDERER = VideoCore IV HW
[notice ] ofAppEGLWindow: createSurface(): GL_VERSION  = OpenGL ES-CM 1.1
[notice ] ofAppEGLWindow: createSurface(): GL_VENDOR   = Broadcom
[notice ] ofAppEGLWindow: createSurface(): -------------
[notice ] ofAppEGLWindow: setupPeripherals(): peripheral setup complete
[notice ] ofAppEGLWindow: setupNativeUDev(): created udev object
[notice ] ofAppEGLWindow: setupMouse(): mouse_fd= 15 devicePath=/dev/input/by-path/platform-3f980000.usb-usb-0:1.2:1.0-event-mouse
[notice ] ofAppEGLWindow: setupMouse(): mouse device name = PixArt USB Optical Mouse
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard_fd= 16 devicePath=/dev/input/by-path/platform-3f980000.usb-usb-0:1.4:1.0-event-kbd
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard device name = Microsoft Wired Keyboard 600
[notice ] ofAppEGLWindow: setupPeripherals(): native event setup complete
[notice ] ofAppEGLWindow: destroyNativeKeyboard()
[notice ] ofAppEGLWindow: destroySurface(): destroying EGL surface

PRITIAEC10, 2017-02-22:
Can you share the config.json you are using, because i think it may be a configuration issue

DHaylock, 2017-02-22:
Actually the config we use is in the repo. Could you send me your Footfall folder? including config.

PRITIAEC10, 2017-02-22:
[Footfall.tar.gz](https://github.com/WatershedArts/Footfall/files/793153/Footfall.tar.gz)

DHaylock, 2017-02-22:
cheers, whilst i'm sorting out a build environment for RPI3 could you check your memory allocation. Try setting it to 128

PRITIAEC10, 2017-02-22:
it is set to 128

DHaylock, 2017-02-22:
Hi @PRITIAEC10,

Just ran your app. Yep there it is no screen data. Going to investigate now.

DHaylock, 2017-02-22:
Hi @PRITIAEC10,

Have just cleaned the target.

[Pr_Footfall.zip](https://github.com/WatershedArts/Footfall/files/793733/Pr_Footfall.zip)

Could you try running ./Pr_Footfall

Best

gundrygs, 2017-02-22:
> I had to do the libs to old-libs trick and use the VS2012-static branch of OfxCv to get it all to compile

I think the correct branch of ofxCv to work against of_v0.9.8_linuxarmv6l_release is 'stable', not 'VS2012-static' - may not make a difference, but it's another thing to try.

DHaylock, 2017-02-22:
Hi @d000000h and @PRITIAEC10,

Because, there is no Notice Log for the config file it suggests that the ofApp.cpp file has been compiled before the source code for footfall has been added. 

I've updated the [compiling.md](https://github.com/WatershedArts/Footfall/blob/master/docs/compiling.md) with clearer instructions.

PRITIAEC10, 2017-02-23:
Hi @DHaylock ,
Thanks a ton, i got the video output now, Trying to configure and compile on my desktop for accuracy , will let you know if i need help. 

Thanks!

d000000h, 2017-02-23:
I went wrong here 👍 
```
Copy all of the Source files into the folder.

cp ../../../../Footfall/Footfall/src/* .

```
Somehow it didnt copy the src files and that was causing the white screen (i think, im running into other problems now)

for me it had to be 👍 
```
Copy all of the Source files into the folder.

cp ../../../Footfall/Footfall/src/* .

```
This can be closed, since the problem was human error ... 

the problem im running into now is the MMAl problem which tells me to remove the libs folder from the ofxCvPicam but when i do that i still keep the problem...

some more fiddling around i guess

DHaylock, 2017-02-23:
Hi @d000000h,

No worries. I updated the docs. 

Open a new Issue and ill have a look.

closed 2017-02-23 by DHaylock