WatershedArts/Footfall #60: Need help in configuring the in and out parameters.

PRITIAEC10, 2017-02-23:
I am trying to configure the config.json file to work for my store. I have recorded a one minute long video[output.mp4] and am trying to count the number of people in and out using that, but the blob always seems to wide, it is ether missing people or counting them more. Can you please check and let me know where i am making the mistake? I have added both the output video and config.json file. THanks in advance!
[outputvideo.tar.gz](https://github.com/WatershedArts/Footfall/files/796374/outputvideo.tar.gz)

DHaylock, 2017-02-23:
Hi @PRITIAEC10,

Ok reflective floors and openCv ... not a good mix.
So the lights are interfering with the MOG, thats why getting the correct parameters is hard. 
You'll see the screen flash as the camera tries to correct the white balance.

You'll first need to try disabling the AutoWhiteBalancing or the exposure settings.
Try 
<pre>raspivid -w 320 -h 240 -fps 25 -awb 0 -t 60000 -o testRecording.h264</pre>
or
<pre>raspivid -w 320 -h 240 -fps 25 -ex 1 -t 60000 -o testRecording.h264</pre>

If this helps you'll need to implement this in the setup.
<pre>cam.setAWBMode((MMAL_PARAM_AWBMODE_T)0);</pre>

These settings get some results until you have corrected the white balancing.
Leave the rest and alter these.
<pre>
"history": 250
"threshold": 50
"mogthreshold": 250
"minarea":20
"maxarea":200
"maxdistance":120
"minsizeone": 55
"minsizetwo": 175
"minsizethree": 200
</pre>

PRITIAEC10, 2017-02-24:
Thanks a lot, if people are moving very fast across the line, it does not count. Which parameter might be affecting?

DHaylock, 2017-02-24:
Hi @PRITIAEC10 

Try eroding the image more so set it to 5.
Change the threshold to 150.
mogThreshold to 75
history to 100.

This is the result I had.

[image]

PRITIAEC10, 2017-02-25:
Are you talking about the tracking threshold or cameraconfig

DHaylock, 2017-02-26:
cameraconfig

closed 2017-02-27 by DHaylock