WatershedArts/Footfall #61: MMAL_PORT_T*, MMAL_BUFFER_HEADER_T RPi2 no libs folder in ofxCvPiCam

d000000h, 2017-02-24:
I am getting the MMAL error even when i have removed the libs folder from ofxCvPiCam directory.

Did have Threadedfootfall running before this
Raspberry Pi2B+ of_v0.9.8
Raspbian fromNOOBS all up to date
I can compile the examples in ofxCvPiCam

I cant compile footfall MMAL error

http://pastebin.com/JiBi8qt4

DHaylock, 2017-02-24:
Hi @d000000h,

No. MMAL is fine.
This is your error, which is odd and is fine on everybuild so far.
<pre>/home/pi/of_v0.9.8_linuxarmv6l_release/apps/myApps/Footfall/src/CameraManager.cpp:95:34: error: invalid initialization of reference of type ‘cv::OutputArray {aka const cv::_OutputArray&}’ from expression of type ‘int’
   GaussianBlur(videoMatrix, _blur);</pre>

Try sticking ofxCv:: infront GaussianBlur?

d000000h, 2017-02-25:
Ive did what you asked;

i changed GaussianBlur to  ofxCv::GaussianBlur in CameraManager.cpp

this is the errors that are being generated
http://pastebin.com/5tYktKgV

if you want i can give you ssh/vnc access to this pi to check it out; or just give me your suggestions :x

DHaylock, 2017-02-26:
Hi @d000000h,

Try.
<pre>
GaussianBlur(videoMatrix,videoMatrix,_blur);
GaussianBlur(_processedMOG,_processedMOG,_blur*2);
</pre>

Best

d000000h, 2017-02-26:
Tried it; give me 👍 
error: ‘_processedMOG’ was not declared in this scope

here is the full buildlog

http://pastebin.com/8QfdHp9a

DHaylock, 2017-02-26:
It's processedMOG. Not _processedMOG. Typo 👍

closed 2017-02-27 by DHaylock