WatershedArts/Footfall #65: If two people walk side by side or one after the other,, which parameter should I adjust?

Hardold, 2017-02-27:


DHaylock, 2017-02-27:
Hi @Hardold,

Are they recording as one person or two?

Best

Hardold, 2017-02-27:
They are recording as one person.I'm confused .Which parameters should I adjust?

Best

PRITIAEC10, 2017-02-27:
Blobsize is to be adjusted

DHaylock, 2017-02-27:
If you have two people walking in side by side, footfall for the time being cant segment them. So i would adjust the maxblobsize to 3/4 the size of the camerawidth. Adjust the midblobsize variable to 1.5x size of minblobsize.

closed 2017-02-27 by DHaylock

Hardold, 2017-02-28:
Hi,
@DHaylock 
  What parameters to adjust maxblobsize, midblobsize and minblobsize? Thinks a lot ! 
`"minsizeone": 10,
"minsizetwo":30,
"minsizethree": 150`?

DHaylock, 2017-02-28:
Tracking config: minblobsize and maxblobsize for the contour finder. min,mid,max size for the event reporting is if the blob can be considered to be bigger than one person.

Hardold, 2017-02-28:
minblobsize is "minarea"?
maxblobsize is"maxarea"?
What is the relationship between( "minsizeone", "minsizetwo", "minsizethree" )and( "minarea" and "maxarea") ?  My picture size is 640*480. How to set them?

DHaylock, 2017-02-28:
Hi @Hardold,

Yes.

The [Configuration.md](./docs/configuration.md) should explain all the parameters.

Best