WatershedArts/Footfall #67: RPi2 not compiling -> error: could not convert ‘((CameraManager*)this)->CameraManager::_blur’ from ‘int’ to ‘cv::Size’

d000000h, 2017-02-27:
RPi2 still not compatible

here is error even after editing CameraManager.cpp

my edit in CameraManager.cpp
`                ofxCv::GaussianBlur(videoMatrix,videoMatrix,_blur);
                // Process the MOG
                if (_useMask)
                {
                        videoMatrix.copyTo(combinedMask, mask);
                        pMOG2->operator()(combinedMask,processedMog);
                }
                else
                {
                        pMOG2->operator()(videoMatrix,processedMog);
                }

                // Get the background image
                pMOG2->getBackgroundImage(background);

                // If you want to see the effect of the shadow ratio threshold. Set the showshadow config to true
                // Copy the MOG's output before processing the mat

                if(_showShadows) copy(processedMog,unprocessed_MOG);

                // Image processing
                threshold(processedMog,_threshold);
                blur(processedMog,_blur);
                dilate(processedMog,_dilateAmount);
                erode(processedMog,_erodeAmount);
                dilate(processedMog,_dilateAmount);
                //ofxCv::GaussianBlur(processedMog, _blur*2);
                ofxCv::GaussianBlur(processedMog,processedMog,_blur*2);
`
here is full build log:

http://pastebin.com/sN7bE9vd

DHaylock, 2017-02-27:
Hi @d000000h,

Try to use the native cv Blur. 
GaussianBlur(processedMog, processedMog, cv::Size(_blur*2,_blur*2), 0);

I have no idea why this isn't working for you. 
Have you checked out the origin/stable branch of ofxCv?

Only thing I can think of is there are conflicting builds of openCv, as ofxCv uses the same method.

d000000h, 2017-02-27:
Thanks a lot for your help, i reversed back to the original posted version you have here (without my edits) changed for the stable branch of ofxCv and now its working fine :) @ 25 fps wow

mayb add this for the RPi2 in the readme or something for other people :dancing_men: 

thanks a lot for your speedy responses .. Are you available for commercial help if we need it with implementing this in our store ? 

thanks again !

closed 2017-02-27 by DHaylock