WatershedArts/Footfall #73: Compiling with windows (msys2)

Unknown212, 2017-03-07:
Hi 
i am quite new to this but I have successfully compiled and run the footfall on Rpi2.  Now I would like to use the desktop for the configuration of the video parameters.  Is it the same procedure when using msys2?  I have compiled openframeworks on my desktop already using msys2 and it can run the examples.  But I cant compile the footfall on the desktop. I am getting the following error:

`In file included from C:/msys64/of_msys/libs/openFrameworks/video/ofDirectShowGrabber.h:9:0,
                 from C:/msys64/of_msys/libs/openFrameworks/video/ofVideoGrabber.h:31,
                 from C:/msys64/of_msys/libs/openFrameworks/ofMain.h:96,
                 from C:/msys64/of_msys/apps/myApps/Footfall/src/CameraManager.h:11,
                 from C:/msys64/of_msys/apps/myApps/Footfall/src/CameraManager.cpp:7:
C:/msys64/of_msys/libs/videoInput/include/videoInput.h:40:0: warning: ignoring #pragma comment  [-Wunknown-pragmas]
 #pragma comment(lib,"Strmiids.lib")

In file included from C:/msys64/of_msys/addons/ofxCvPiCam/libs/mmal/src/interface/mmal/mmal.h:363:0,
                 from C:/msys64/of_msys/addons/ofxCvPiCam/src/ofxCvPiCam.h:12,
                 from C:/msys64/of_msys/apps/myApps/Footfall/src/CameraManager.h:14,
                 from C:/msys64/of_msys/apps/myApps/Footfall/src/CameraManager.cpp:7:
C:/msys64/of_msys/addons/ofxCvPiCam/libs/mmal/src/interface/mmal/mmal_common.h:40:33: fatal error: interface/vcos/vcos.h: No such file or directory
 #include <interface/vcos/vcos.h>
                                 ^
compilation terminated.
make[1]: *** [/c/msys64/of_msys/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:202: obj/msys2/Release/src/CameraManager.o] Error 1
make[1]: Leaving directory '/c/msys64/of_msys/apps/myApps/Footfall'
make: *** [/c/msys64/of_msys/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:130: Release] Error 2
`

franmaguire, 2017-03-07:
Hi @Unknown212, 
In the configuration.md,  
> On your desktop compile the footfall app. 

This is still on your raspberrypi.  Think of that text as a wannabe heading that didnt make it out of the paragraph.  If you re read configuration.md it then makes sense.

franmaguire, 2017-03-07:
Also the line

> Commenting USE_PI_CAM and uncommenting USE_VIDEO

is in the same file that the next step asks you to edit.   Footfall/src/CameraManager.cpp, line 58 - 62

DHaylock, 2017-03-08:
Cheers @franmaguire 

Hope that helps @Unknown212.

I've updated the ReadMe to reflect this issue

Best

closed 2017-03-08 by DHaylock