WatershedArts/Footfall #74: Error in runtime

maxso96, 2017-03-08:
Hello I'm new here and I've been trying to run this on my Raspberry Pi 3.
I already finished compiling footfall as directed here:
[http://www.oodlestechnologies.com/blogs/People-count-by-raspberry-pi-3-using-openframework#disqus_thread](url)

However I'm experiencing a problem during runtime saying

-------------------- Footfall --------------------
Loading Configuration - Configuration File Found!
Setting Up Camera Manager - Using Pi Camera
Setting Up Tracking Manager
OpenCV Error: Assertion failed (ksize.width > 0 && ksize.width % 2 == 1 && ksize.height > 0 && ksize.height % 2 == 1) in createGaussianFilter, file /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/smooth.cpp, line 816
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-ISmtkH/opencv-2.4.9.1+dfsg/modules/imgproc/src/smooth.cpp:816: error: (-215) ksize.width > 0 && ksize.width % 2 == 1 && ksize.height > 0 && ksize.height % 2 == 1 in function createGaussianFilter

[notice ] ofAppEGLWindow: destroyNativeKeyboard()
[notice ] ofAppEGLWindow: destroyNativeKeyboard(): unable to reset terminal
[notice ] ofAppEGLWindow: destroySurface(): destroying EGL surface
Aborted

Can you please help me. Thanks my RPi is with linuxarm7vl and of version v.0.9.0

DHaylock, 2017-03-08:
Hi @maxso96 

What is you blur value?

Best

maxso96, 2017-03-08:
`		cv::GaussianBlur(videoMatrix,videoMatrix, cv::Size(_blur,_blur), 0);

		cv::GaussianBlur(processedMog, processedMog, cv::Size(_blur*2,_blur*2), 0);`

I think the problem is here. My config.json is set on default values. And I dont know what is a mask. I just edited the CameraManager.cpp 

Regards

DHaylock, 2017-03-08:
Hi @maxso96,

Interesting.

Can you try the ofxCv GaussianBlur and just add an odd number into the parameters

<code>GaussianBlur(processedMog,5); </code>

Best

maxso96, 2017-03-08:
It still doesn't work. I looked on issue #67 before and it fixed the compiling error. 
But now I still can't run the program. Here's my edited code, is this still wrong?

`ofxCv::GaussianBlur(videoMatrix,videoMatrix, cv::Size(_blur,_blur), 5);
ofxCv::GaussianBlur(processedMog, processedMog, cv::Size(_blur*2,_blur*2), 5);`

Other than this file and config.json I didn't edit other files. So I suppose that the error is on here. I also didn't install the OpenCV library as I read that it might conflict with ofxCv.

Thanks again!

DHaylock, 2017-03-08:
Hi @maxso96 

Did you checkout origin/stable of ofxCv?

Best

maxso96, 2017-03-08:
Hello Sir @DHaylock 
I already installed the stable branch of ofxCv, but when I try to compile it goes on error.

`ofxCv::GaussianBlur(videoMatrix,videoMatrix, cv::Size(_blur,_blur), 5); ofxCv::GaussianBlur(processedMog, processedMog, cv::Size(_blur*2,_blur*2), 5);`

When I omit the ofxCV:: from the front it will compile, but it will throw the assertion error on runtime.

DHaylock, 2017-03-08:
Hi @maxso96,

Could you try this again?
Wondering whether the standard deviation parameter is doing something odd.

<code>GaussianBlur(videoMatrix, _blur);
GaussianBlur(processedMog, _blur*2);</code>

Best

maxso96, 2017-03-08:
Wooh! It finally worked after a long compile time! May I ask what's the problem with the original code I issued here? In hindsight, I should've used the stable branch instead of the static one.

Thank you so much for your help Sir @DHaylock!
Best wishes on your future projects and on this as well 👍

DHaylock, 2017-03-08:
Hi @maxso96,

No worries.

Only thing I can think of is a conflict with the libs.
The origin/stable branch is ... well stable :) 
Other than that I think that the final parameter was the issue.

If you look in the [filtering docs](http://docs.opencv.org/2.4/modules/imgproc/doc/filtering.html?highlight=gaussianblur#gaussianblur) its the sigmaX,sigmaY values.

Best of luck

David

closed 2017-03-08 by DHaylock

maxso96, 2017-03-08:
Another question Sir @DHaylock 

When I ran footfall I noticed that it can't connect to the db. I followed exactly what is written there, but it says
[ error ] Can't upload due to no network

Is there something wrong with my config.json when I set
"postserver":"http://localhost", 
"postextension":"upload.php"