WatershedArts/Footfall #75: Network upload problems

maxso96, 2017-03-08:
I got to finally run the program without experiencing error on the compile and runtime of the program itself. I tried to edit the config.json file to connect it to the db.
`"usehttp": true,

"postserver":"http://localhost",
"postextension":"upload.php",`

But when I ran footfall I noticed that it can't connect to the db. I followed exactly what is written in the server.md, but it says

`[warning] ofThread: - name: Thread 2 - Calling startThread with verbose is deprecated.
 - Not Keeping Backup Logs
[ error ] Can't upload due to no network`

What seems to be causing this? Please help thanks

maxso96, 2017-03-08:
Yes I get the page but its static.

DHaylock, 2017-03-08:
Ok, here come a stream of silly questions. :) can you check that your secret keys match?

maxso96, 2017-03-08:
The key to dbconnect.php is the same, do I have to also edit the secret.php?

DHaylock, 2017-03-08:
Hi @maxso96 

[This](https://github.com/WatershedArts/Footfall/blob/master/Web/includes/secret.php#L2) line should match your secret key in your config.json. It's like another level of protection.

maxso96, 2017-03-08:
Wew silly of me :) I matched them both just like what you said.
Thanks again Sir, now I get it and it's operating completely!

Again, best wishes and thanks a lot! 💯

DHaylock, 2017-03-08:
No worries. I will update the docs as this isn't in them, sorry about that.

Best

closed 2017-03-08 by DHaylock