WatershedArts/Footfall #76: run it on windows

hardjob, 2017-03-08:
hello i want to try this project on my desktop before  deploying it to Rpi, how can i install openframework.., should i just include those missing files or what (ofMain.h, ofxCv.h... )

DHaylock, 2017-03-09:
Hi @hardjob 

I don't have access to windows hardware. However, if you follow the guide on the [openFrameworks site](http://openframeworks.cc/setup/vs/). You be able to install and compile it. 
As for Footfall. You'll need to clone the repos from the getrepos.sh file.
I haven't built a windows version but am sure it will work.

Best

closed 2017-03-09 by DHaylock

gundrygs, 2017-03-10:
Hi @hardjob  - I have built it on Windows 10 with Visual Studio 2015 Community Edition - it should work on other versions with minor adjustments

 @DHaylock - I hope it's OK if I reproduce my notes below:

**Build Waterfall Footfall on Windows 10 with Visual Studio**

1. Using VisualStudio 2015 Community edition, follow the instructions at http://openframeworks.cc/setup/vs/
2. Download and unzip VisualStudio OpenFrameworks code from here http://openframeworks.cc/download/
This will create a directory %HOMEPATH%\Downloads\of_v0.9.8_vs_release\of_v0.9.8_vs_release

3. Install Git for Windows: https://git-for-windows.github.io/ if you don't already have it.
4. Then to get the Footfall code from Windows shell:

_cd %HOMEPATH%
git clone https://github.com/WatershedArts/Footfall.git
set OF=%HOMEPATH%\Downloads\of_v0.9.8_vs_release\of_v0.9.8_vs_release
cd %OF%\addons
git clone https://github.com/kylemcdonald/ofxCv **-b stable**
git clone https://github.com/arturoc/ofxHttpUtils
git clone https://github.com/jefftimesten/ofxJSON
git clone https://github.com/paulvollmer/ofxCsv_

5. Create a new project in VS called Footfall using the openFrameworks plugin you installed at Step 1 – do NOT check the box to create a directory

Choose the following Official addons when prompted:
ofxNetwork
ofxOpenCv

Choose the following Community addons when prompted:
ofxCsv
ofxCv
ofxHttpUtils
ofxJSON

6. From the Windows shell:

_cd %HOMEPATH%
robocopy Footfall\Footfall\   %OF%\apps\myApps\ -e_

Now go back to the Solution Explorer pane in VS, right click on Footfall -> src, and use Add -> Existing Item... to highlight all files in apps\myApps\Footfall\src and add them to the Project

7. Amend CameraManager.h as shown below, #defining USE_VIDEO:

//#define USE_PI_CAM
//#define USE_WEBCAM
#define USE_VIDEO

To specify a video file to run, follow David's configuration guide here https://github.com/WatershedArts/Footfall/blob/master/docs/configuration.md

8. There are two apparent bugs that we need to work around :-(

Firstly, the openFrameworks plugin doesn't seem to REALLY add ofxOpenCv to the dependencies, and this leads to many link errors - I had to remove and re-add ofxOpenCv (right click on Footfall in the Solution Explorer pane of VS -> openFrameworks addons -> uncheck ofxOpenCv -> press OK, then repeat but check the ofxOpenCv tickbox -> press OK)

Secondly, there is an additional bug in the openFrameworks plugin where the Additional Dependencies directory list gets corrupted with control characters and causes a single link failure - https://github.com/openframeworks/openFrameworks/issues/4724
To work around this, right click on the Footfall project in the Solution Explorer pane in VS and choose Properties. Select Linker->Input and replace any instances of %3b with semicolon space. There are quite a few, and you may need to do it for both Debug and Release targets, so this is pretty painful …

9. Build the solution - it will take about 10 minutes to build openFrameworks, then the addons, and finally Footfall itself

hardjob, 2017-03-20:
i got some errors related to ofxCv :
cannot open source file "ofxCv/Calibration.h"
....
i tried to remove and re-add ofxCv from OF addons but still the same error

gundrygs, 2017-03-20:
@hardjob,

So, do you have something that looks like this in the Solution Explorer ?
[image]

hardjob, 2017-03-20:
I fixed that by editing include directories
[image]

hardjob, 2017-03-22:
@gundrygs  i build it and run it , but i got a blank window, and the debugger don't show any error : 
[image]
i realised that vs can't cout a non-consol application ....
how can i deal with that ??

hardjob, 2017-03-22:
@gundrygs can you please send me the video sample you use

gundrygs, 2017-03-23:
Hi hardjob,

 

Try the attached – you will need to alter the filename at line 47 in CameraManager.cpp to “testFeb20pm.mp4”

 

I think you will also need to install the video codec pack as mentioned at http://openframeworks.cc/setup/vs/ - link is http://filehippo.com/download_klite_codec_pack/67445/

 

Regards,

 

George

hardjob, 2017-03-24:
i didn't receive the attachment , please send it at thepappillon@gmail.com
also i often get this message , i don't knwo why 
[image]

gundrygs, 2017-03-24:
Hi hardjob – attachment sent – I’m afraid I can’t help with the exception – remote debugging is pretty impossible !

 

Hope you work it out,

 

Regards,

 

George

 <https://cloud.githubusercontent.com/assets/8091800/24275757/c6060ef8-1029-11e7-9e30-0292fe2e1025.PNG>

randomly123, 2017-08-16:
On Visual studio 2015, I was finally able to build the project without any errors . I ran it also using Alt F5, which is when I got the below screenshot. However, I am unable to see the video and the line and the square for cutting the Footfall/the crossover line, layered on top of the video that I usually see on the raspberry Pi. In fact I am unable to see the video being played back at all on Windows.

For building it on Windows, I am using YOURTESTFOOTAGE.mp4 on  Line number 47 - as mentioned by grundygs

the OF\apps\myApps\Footfall\bin\data folder contains, the standard two files - "config.json" and "yourtestfootage.mp4"

There are no build errors, its just that I cant see the video and the cutting line. Neither do I get any output of FF +1 or -1. 
 Only the CMD window in black and a standard blank, empty, VS forms Window.

Thank you, for the pointers,

randomly123, 2017-08-21:
Anyways..now I am able to run the FF on a video, however, the configuration of the damn things is so tough!!!