WatershedArts/Footfall #77: Rotate Tracking Area?

CrossPT, 2017-03-15:
Hi.
First of all thanks for this amazing code!
I'm currently testing the footfall of this area 
[image]

I want to count the footfall on people (rotate the tracking bar) without changing the position of the RPi, like this:
[image]

Is it possible?

DHaylock, 2017-03-15:
Hi @CrossPT,

Yeah totally possible though you will need to alter the source code. I haven't had time to implement this properly. But for now. Compile these files. Essentially in your config.json. The code will look for a 0 in either of the startposition parameters. If it finds 0 in the y it will flip the orientation of tracking. Just tried it on my machine seems to be ok may want to refine it abit.

[Archive.zip](https://github.com/WatershedArts/Footfall/files/845221/Archive.zip)

Hope it helps.

Best

CrossPT, 2017-03-15:
Thumbs up for the quick reply! 👍 
Will test and reply with the changes

CrossPT, 2017-03-16:
Works like a a charm!

Maybe make a small change in code to flip the orientation of tracking. The startposition parameters may be arranged to one parameter only.

Something like this in config.json:
startpos: <value>,
landscape: <true/false>

closed 2017-03-16 by CrossPT