WatershedArts/Footfall #82: Compile errors for ofxCv "error: 'glm' does not name a type"

FelixEder, 2017-04-05:
I have a problem with ofxCv when trying to compile the program.

I'm using a Raspberry Pi 3 Model B with Raspbian Jessie. and using the OpenFrameWorks 0.9.8 as well as ofxCv stable branch.

When I try to compile it, I get the error "error: 'glm' does not name a type". I get this at least 20 times during one compilation.

How do I solve this error?

mcarlier, 2017-04-06:
Hi!
I set up the project yesterday with the same config as yours.
I needed to go to the ofxCv page : https://github.com/kylemcdonald/ofxCv
and replace manually the ofxCv file by the stable branch in my openFramework file. 
Did you did it ?

FelixEder, 2017-04-06:
I did manually replace ofxCv in the addons folder with the stable version from that repo you linked. I got the same issues still. What do you mean by replacing the "ofxCv file"? I replaced the whole folder, is there instead a specific file that should be replaced and not the whole directory?

Thanks for the help!

mcarlier, 2017-04-06:
No, I also replace the all folder, like you did. 
Are you sure of your openframework version ?

DHaylock, 2017-04-06:
Hi Both,

Navigate to the ofxCv folder in your addons folder.
Inside do the following command.

<code>git checkout origin/stable</code>

This will checkout the correct branch.

[Issue 62](https://github.com/WatershedArts/Footfall/issues/62)

FelixEder, 2017-04-06:
@mcarlier I used this guide to install OFW on my Raspberry:
 http://openframeworks.cc/setup/raspberrypi/raspberry-pi-getting-started/
I used the same version as stated in the guide, 0.9.8.

@DHaylock I tried it out and it worked, thanks! The glm-error doesn't appear anymore, but now I get a different error:
`openFrameworks/apps/Footfall/Footfall/src/CameraManager.cpp:24:8: error: no match for ‘operator=’ (operand types are ‘cv::Mat’ and ‘CvMat*’)
  mask = cvCreateMat(_cameraConfig.cameraheight, _cameraConfig.camerawidth, CV_8UC1);`

FelixEder, 2017-04-12:
I managed to solve the issue myself by changing my installed OpenCV version from 3.2.0 to 2.4.9 and that managed to fix it.

@DHaylock Maybe the OpenCV version used in Footfall should be more clearly stated on the README? Would have made my install a lot easier.

closed 2017-04-12 by FelixEder

Ferni492, 2018-08-07:
@felixeder how did you change the opencv versión??