WatershedArts/Footfall #83: Errors in CameraManager.cpp; invalid cast and no member named setDouble

emilelmarsson, 2017-04-06:
I'm using:
Raspberry Pi 3b
OS: Raspbian Jessie.
openFrameworks version: 0.9.8 armv6l
ofxCv: origin/stable branch
OpenCV version: 3.2.0

When compiling the Footfall repository with `sudo make -j3` I get these errors in the CameraManager.cpp file:
`/home/pi/openFrameworks/apps/Footfall/Footfall/src/CameraManager.cpp:41:12: error: invalid cast to abstract class type ‘cv::BackgroundSubtractorMOG2’
            );`

`/home/pi/openFrameworks/apps/Footfall/Footfall/src/CameraManager.cpp:43:9: error: ‘class cv::BackgroundSubtractorMOG2’ has no member named ‘setDouble’
  pMOG2->setDouble("fTau", _cameraConfig.shadowPixelRatio);`

I followed the guide. There shouldn't be anything wrong with the code so how could I get these errors?

DHaylock, 2017-04-24:
Have you installed a different version of OpenCv?

It looks like the version of OpenCv is incompatible with oF's Libs.

emilelmarsson, 2017-04-24:
Yeah. I changed the OpenCV version to 2.4.9 and thereby got it working. I just forgot to close the issue. Sorry about that.

closed 2017-04-24 by emilelmarsson