WatershedArts/Footfall #91: Not able to make Footfall

reddy6980, 2017-04-24:
Hi,

When making the file we are getting below error

make[1]: *** [obj/linuxarmv6l/Release/src/CameraManager.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/Footfall/Footfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2

Can you please help us in solving the error.

Thanks,
Vamsi

anthonysuterag, 2017-04-24:
Hi,

I also experienced the same error.

Please let me know if you manage to solve the error.
Thank you

Anthony.

DHaylock, 2017-04-24:
Hi Both,

Can you use copy the log into something like Pastebin?
The Error 1 symbol only tells me it's something with the CameraManager file.

Best

reddy6980, 2017-04-24:
Hi,

Was able to solve it by downloading add-ons like ofxCv,ofxCsv,ofxOpenCv,
ofxJSON,ofxHttpUtils,ofxCvPiCam

Now i am able to make the file but the make stops at the following error
and remains in the same position without completing

In file included from
/home/pi/openFrameworks/apps/Footfall/Footfall/src/TrackingManager.h:15:0,
                from
/home/pi/openFrameworks/apps/Footfall/Footfall/src/ofApp.h:12,
                from
/home/pi/openFrameworks/apps/Footfall/Footfall/src/ofApp.cpp:8:
/home/pi/openFrameworks/apps/Footfall/Footfall/src/TrackingHistory.h:
In member function ‘void TrackingHistory::draw(int, int)’:
/home/pi/openFrameworks/apps/Footfall/Footfall/src/TrackingHistory.h:90:21:
warning: comparison between signed and unsigned integer expressions
[-Wsign-compare]
   for(int i = 0; i < trackingHistory.size(); i++)

Pls let me know what the error is.

Regards,
Vamsidhara Reddy V

anthonysuterag, 2017-04-24:
Hi,

This is the error that I got when I run "sudo make -j3" 
https://pastebin.com/nXGTFAz6

I already put ofxCv,ofxCsv,ofxOpenCv,ofxJSON,ofxHttpUtils,ofxCvPiCam in my addons.

And for the information: I'm using Raspberry Pi 3 model B and openFrameworks version 0.9.8 for linux armv6

Thank you in advance.

Anthony.

DHaylock, 2017-04-24:
Hi 

You need to checkout a different branch of ofxCv origin/stable should be the correct one.

[Issue 82](https://github.com/WatershedArts/Footfall/issues/82)

Best

reddy6980, 2017-04-24:
Hi,

I am able to make the file and able to run it also. The database is also updating but i wanted to know when we run the footfall.sh file can we use any command that displays the video also while executing the command.

I am connecting using vnc viewer to the raspberry pi.

anthonysuterag, 2017-04-24:
Hello, 

I already change the branch of my ofxCv.

But now, I get the following error: https://pastebin.com/6hFs8AxE

Do I need to change my openCV version for that problem?

Anthony

tsjose, 2017-05-02:
I'm also getting this error. any update on how to fix it?

Here is my log

`In file included from /home/pi/openFrameworks/apps/Footfall/Footfall/src/TrackingManager.h:15:0,
                 from /home/pi/openFrameworks/apps/Footfall/Footfall/src/ofApp.h:12,
                 from /home/pi/openFrameworks/apps/Footfall/Footfall/src/ofApp.cpp:8:
/home/pi/openFrameworks/apps/Footfall/Footfall/src/TrackingHistory.h: In member function ‘void TrackingHistory::draw(int, int)’:
/home/pi/openFrameworks/apps/Footfall/Footfall/src/TrackingHistory.h:90:44: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    for(int i = 0; i < trackingHistory.size(); i++)
                                            ^
In file included from /home/pi/openFrameworks/apps/Footfall/Footfall/src/CameraManager.h:20:0,
                 from /home/pi/openFrameworks/apps/Footfall/Footfall/src/CameraManager.cpp:7:
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h: At global scope:
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h:200:13: warning: ‘void color_callback(MMAL_PORT_T*, MMAL_BUFFER_HEADER_T*)’ declared ‘static’ but never defined [-Wunused-function]
 static void color_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h:201:13: warning: ‘void gray_callback(MMAL_PORT_T*, MMAL_BUFFER_HEADER_T*)’ declared ‘static’ but never defined [-Wunused-function]
 static void gray_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:200: recipe for target 'obj/linuxarmv6l/Release/src/CameraManager.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/CameraManager.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from /home/pi/openFrameworks/apps/Footfall/Footfall/src/CameraManager.h:20:0,
                 from /home/pi/openFrameworks/apps/Footfall/Footfall/src/ofApp.h:11,
                 from /home/pi/openFrameworks/apps/Footfall/Footfall/src/ofApp.cpp:8:
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h: At global scope:
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h:200:13: warning: ‘void color_callback(MMAL_PORT_T*, MMAL_BUFFER_HEADER_T*)’ declared ‘static’ but never defined [-Wunused-function]
 static void color_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.h:201:13: warning: ‘void gray_callback(MMAL_PORT_T*, MMAL_BUFFER_HEADER_T*)’ declared ‘static’ but never defined [-Wunused-function]
 static void gray_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^
make[1]: Leaving directory '/home/pi/openFrameworks/apps/Footfall/Footfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2`

xvii, 2017-06-22:
Hello,

I also have the same problem, when try to make it.

```
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl gtk+-3.0 
/bin/sh: 1: ./Footfall: not found
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:176: recipe for target 'RunRelease' failed
make: *** [RunRelease] Error 127

```

closed 2017-12-12 by DHaylock