WatershedArts/Footfall #93: Compiling Errors, armv6l, Jessie, RPi 3

screwu666, 2017-05-22:
Hello, thanks for this app in the first place.

Trying to compile on RPi 3, got linux Jessie and trying, have already compiled armv6l OF and trying to compile the app, but keep running into this error. Do you think there is a way to fix this?

"""

make[1]: *** No rule to make target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCv/./src/CameraManager.o', needed by 'bin/Footfall'.  Stop.
make[1]: *** Waiting for unfinished jobs....
cat: obj/linuxarmv6l/Release/.compiler_flags: No such file or directory
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/Footfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2

"""

screwu666, 2017-05-22:
noticed that updating manually new ofxCv, ofxCsv in the OF folder makes my compilation advance a bit and now he asks for ofxJSON, therefore, I assume that i'll report once i have satisfied all the compilers needs and see if it compiles. Probably something with the file makefile that i wrote...

screwu666, 2017-05-22:
Got down to this

 typedef void (*MMAL_PORT_CLOCK_EVENT_CB)(MMAL_PORT_T *port, const MMAL_CLOCK_PAYLOAD_T *event);
                                                                   ^
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:361: recipe for target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCvPiCam/libs/mmal/src/interface/mmal/util/mmal_graph.o' failed
make[1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCvPiCam/libs/mmal/src/interface/mmal/util/mmal_graph.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.cpp: In member function ‘void ofxCvPiCam::setup(int, int, bool)’:
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.cpp:63:188: warning: narrowing conversion of ‘ofxCvPiCam::width’ from ‘int’ to ‘uint32_t {aka unsigned int}’ inside { } [-Wnarrowing]
         MMAL_PARAMETER_CAMERA_CONFIG_T cam_config = {{ MMAL_PARAMETER_CAMERA_CONFIG, sizeof (cam_config)}, width, height, 0, 0,width, height, 3, 0, 1, MMAL_PARAM_TIMESTAMP_MODE_RESET_STC };
                                                                                                                                                                                            ^
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.cpp:63:188: warning: narrowing conversion of ‘ofxCvPiCam::height’ from ‘int’ to ‘uint32_t {aka unsigned int}’ inside { } [-Wnarrowing]
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.cpp:63:188: warning: narrowing conversion of ‘ofxCvPiCam::width’ from ‘int’ to ‘uint32_t {aka unsigned int}’ inside { } [-Wnarrowing]
/home/pi/openFrameworks/addons/ofxCvPiCam/src/ofxCvPiCam.cpp:63:188: warning: narrowing conversion of ‘ofxCvPiCam::height’ from ‘int’ to ‘uint32_t {aka unsigned int}’ inside { } [-Wnarrowing]
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/Footfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2

screwu666, 2017-05-22:
Solved all, thanks !

closed 2017-05-22 by screwu666

xvii, 2017-07-22:
How u solved ?

wassgha, 2017-09-17:
@screwu666 can you please explain how you solved this?