WatershedArts/Footfall #94: Cant make the program

xvii, 2017-06-27:
Hello,

Whe I try to make the Footfall i get that error:

```
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl gtk+-3.0 
/bin/sh: 1: ./Footfall: not found
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:176: recipe for target 'RunRelease' failed
make: *** [RunRelease] Error 127

```

mellolr1, 2017-07-31:
Also getting en error on RPI3:

pi@raspberrypi:~/openFrameworks/apps/Footfall/Footfall $ make -j3
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl gtk+-3.0 
Compiling OF library for Release
make[1]: Entering directory '/home/pi/openFrameworks/libs/openFrameworksCompiled/project'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl gtk+-3.0 
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl gtk+-3.0 
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl gtk+-3.0 
Done!
make[1]: Leaving directory '/home/pi/openFrameworks/libs/openFrameworksCompiled/project'

Compiling Footfall for Release
make[1]: Entering directory '/home/pi/openFrameworks/apps/Footfall/Footfall'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl gtk+-3.0 
Compiling /home/pi/openFrameworks/apps/Footfall/Footfall/src/CameraManager.cpp
g++ -c -O3 -DNDEBUG -Wall -std=c++14 -DGCC_HAS_REGEX -march=armv6 -mfpu=vfp -mfloat-abi=hard -fPIC -ftree-vectorize -Wno-psabi -pipe -DOF_USING_GTK -DOF_USING_GTK -DTARGET_RASPBERRY_PI -DSTANDALONE -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -D_FORTIFY_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM  -I/opt/vc/include -I/opt/vc/include/IL -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -pthread -I/usr/include/gstreamer-1.0 -I/usr/include/AL -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/arm-linux-gnueabihf/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng12 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng12 -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -I/usr/include/opencv -I/home/pi/openFrameworks/libs/fmodex/include -I/home/pi/openFrameworks/libs/glfw/include -I/home/pi/openFrameworks/libs/glfw/include/GLFW -I/home/pi/openFrameworks/libs/kiss/include -I/home/pi/openFrameworks/libs/poco/include -I/home/pi/openFrameworks/libs/tess2/include -I/home/pi/openFrameworks/libs/utf8cpp/include -I/home/pi/openFrameworks/libs/utf8cpp/include/utf8 -I/home/pi/openFrameworks/libs/openFrameworks -I/home/pi/openFrameworks/libs/openFrameworks/utils -I/home/pi/openFrameworks/libs/openFrameworks/types -I/home/pi/openFrameworks/libs/openFrameworks/communication -I/home/pi/openFrameworks/libs/openFrameworks/math -I/home/pi/openFrameworks/libs/openFrameworks/graphics -I/home/pi/openFrameworks/libs/openFrameworks/app -I/home/pi/openFrameworks/libs/openFrameworks/3d -I/home/pi/openFrameworks/libs/openFrameworks/video -I/home/pi/openFrameworks/libs/openFrameworks/sound -I/home/pi/openFrameworks/libs/openFrameworks/gl -I/home/pi/openFrameworks/libs/openFrameworks/events -pthread -I/usr/include/gstreamer-1.0 -I/usr/include/AL -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/arm-linux-gnueabihf/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng12 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng12 -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -I/usr/include/opencv -I/home/pi/openFrameworks/apps/Footfall/Footfall/src -I/home/pi/openFrameworks/addons/ofxCsv/src -I/home/pi/openFrameworks/addons/ofxJSON/src -I/home/pi/openFrameworks/addons/ofxJSON/libs -I/home/pi/openFrameworks/addons/ofxJSON/libs/jsoncpp -I/home/pi/openFrameworks/addons/ofxJSON/libs/jsoncpp/include -I/home/pi/openFrameworks/addons/ofxJSON/libs/jsoncpp/include/json -I/home/pi/openFrameworks/addons/ofxJSON/libs/jsoncpp/src -I/home/pi/openFrameworks/addons/ofxJSON/libs/jsoncpp/include -I/home/pi/openFrameworks/addons/ofxJSON/libs/jsoncpp/include/json -I/home/pi/openFrameworks/addons/ofxCv/libs/ofxCv/include -I/home/pi/openFrameworks/addons/ofxCv/libs/CLD/include/CLD -I/home/pi/openFrameworks/addons/ofxCv/src -Isrc -I/home/pi/openFrameworks/addons/ofxOpenCv/src -I/home/pi/openFrameworks/addons/ofxOpenCv/libs -I/home/pi/openFrameworks/addons/ofxHttpUtils/src -I/home/pi/openFrameworks/addons/ofxCvPiCam/src -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src/interface -I/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src/interface/mma [... cut ...]
 static void gray_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:200: recipe for target 'obj/linuxarmv6l/Release/src/ofApp.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/ofApp.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/Footfall/Footfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2

JamesDale, 2017-09-10:
Same issue here on the pi 3, bunch of issues such as this one:

/home/pi/openFrameworks/addons/ofxCv/libs/ofxCv/src/Utilities.cpp:72:2: error: ‘glm’ does not name a type
  glm::vec2 toOf(Point2f point) {
  ^~~

DHaylock, 2017-12-12:
Hi

You need to make sure you are using the [stable branch ](https://github.com/WatershedArts/Footfall/issues/82)

Best

closed 2017-12-12 by DHaylock