WatershedArts/Footfall #101: Noisy camera input

wassgha, 2017-12-06:
Hello,

I've noticed a lot of false positives in my data and after investigating it, it turns out the background subtraction algorithm isn't working as it's supposed to. The problem is, it's the input image that has noise (just the feed from the camera) which causes background subtraction to fail. However, when using  `raspistill` to preview the camera, the noise (flicker of light/dark as seen on the video) does not exist at all! I am not sure if this has anything to do with the library that is used to access the camera or with footfall itself. Another thing to note is that the colors are "reversed(?)" where reds show up as blue. Although this might not affect detection, it might be related to the flicker issue.

Is there any parameter I could tweak to make this noise go away?

[image]

wassgha, 2017-12-09:
@DHaylock ?

DHaylock, 2017-12-12:
Hi @wassgha 

I have never seen a camera behave like this before. It could be trying to white balance the image?

You could try setting [white balance to manual](https://github.com/orgicus/ofxCvPiCam/blob/master/example-ofxCvPiCam-allSettings/src/testApp.cpp)

Try just running the raspi-vid command and change the [white settings](https://www.raspberrypi.org/documentation/raspbian/applications/camera.md)

Otherwise, in the config.json file make the history variable larger. It should smooth out the image.

David

wassgha, 2017-12-12:
Hello! Thank you very much for the reply! Unfortunately I tried both suggestions but no luck.
Here's what raspistill shows

[image]

which is definitely noisy, but still different from the effect I get in the preview window of Footfall (which seems like some kind of re-focusing)

[image]

It might just be the sensor...

randomly123, 2018-01-20:
Never experienced what you have experienced man. Might just be the RaspiCam.

wassgha, 2018-02-26:
Update on this: Just bought another raspicam (v2.0 this time), still the same problem. Also, to add more details, the colors are not correct on the picture (the blue bench you see there is actually red not blue).

prisikarm, 2019-02-23:
Struggling with the same problem tried different cameras... (Pi3B and Pi0W)

prisikarm, 2019-02-23:
I discovered that this is due to "blur" function. is blur is set to zero, everything is ok, but there will be other problems when blur is zero...