WatershedArts/Footfall #106: make -j3 error

mj222, 2018-01-28:
Compiling Footfall for Release
make[1]: Entering directory '/home/pi/openFrameworks/apps/Footfall/Footfall'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libcurl glfw3 rtaudio gtk+-3.0 libmpg123 
with PKG_CONFIG_LIBDIR=
make[1]: *** No rule to make target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCsv/./src/main.o', needed by 'bin/Footfall'.  Stop.
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/pi/openFrameworks/apps/Footfall/Footfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed
make: *** [Release] Error 2
pi@raspberrypi:~/openFrameworks/apps/Footfall/Footfall $

black23, 2018-07-19:
Same for me, any fix for this?

DHaylock, 2018-08-23:
You need to move the Footfall folder out of Project folder 
```
mv /home/pi/openFrameworks/apps/myApps/Footfall /home/pi
cd Footfall
mv Footfall ../openFrameworks/apps/myApps/
```

closed 2018-08-23 by DHaylock